MarlinFirmware/Marlin #14186: [BUG] No eprom.dat on RE-ARM

kurtisane, 2019-05-30:
### Description
I followed the Guide for Marlin 2.0. 
I'm facing the issue that no eprom.dat is created. Tried 3 SD Cards already with 2 Re-ARM Boards. 
the FIRMARE.CUR gets created successfully and the timestamp is also changing. 
The SD-Cards are all formated to FAT32 and named correctly "REARM". 

>>> M500
SENDING:M500
echo:Settings Stored (629 bytes; crc 15680)

But no eprom.dat
The Re-ARMs are acting kind of randomly. Sometimes the SD-Card can not be mounted after putting the firmare on it and restarting the board.
I dont know if that is related or not. 

I'm on Windows 10 using Marlin 2.0 Build from 29.05.2019. 

### Steps to Reproduce

1.  Format SD-Card to FAT32
2. Insert into Re-ARM and plug in the Re-ARM Board
3. Build Marlin 2.0 using VSC and Platform IO
4. Copy firmware.bin to mounted SD-Card
5. Restart the Re-ARM using the Button on Board
6. Kind of randomly 6.1 or 6.2
6.1. Does not mount the SD-Card anymore
6.1.1 Pronterface can connect and M119 works
6.1.2. M500, M501 works, can't see wether or not eprom.dat exists
6.2. Does mount the SD-Card correctly
6.2.1. Pronterface connects and M119 works
6.2.2. M500, M501 works, but still no eprom.dat

**Expected behavior:** [What you expect to happen]

M500 creates eprom.dat

**Actual behavior:** [What actually happens]

M500 is successfull, but no eprom.dat

p3p, 2019-05-30:
By default Marlin LPC176x builds use Flash EEPROM emulation rather than sdcard emulation so it doesn't create the file, that guide is a little out of date.

boelle, 2019-07-21:
will close this as its mostly a misunderstanding and an outdated guide
we can always reopen

closed 2019-07-21 by boelle