MarlinFirmware/Marlin #14188: TMC2130 weird behaviour during build leveling

SSkull995, 2019-05-30:
Hi everybody! 
I hope this is not a dumb question and that you could help me out. I have a problem that I am facing if I start a print with Cura, on my CreatBot with marlin and TMC2130 on x and y axis. 
When I start a new print, the print head moves to the x0 y0, detecting the obstacle; however, when my Bltouch starts levelling (G29), the x motor seems to push the head to the limit, and I hear the typical rattling of the belt. This is very weird, because it never happened before. 
I use to have sensitivity to 2 for both x and y, but today I had to change them to 5 because the head used to stop way before the end of the rods. 
Ah, when I do G28 and then G29, nothing happens, everything is smooth! 
Do you know what should it be to cause this? I use Cura with OctoPrint.
Thanks a lot in advance and have a nice evening/day

cemalgultekin, 2019-05-31:
well, you can always save your slice in .gcode file and open up with a text editor to see what cura does with your printer. you can copy the related direct output here so that more experienced people might tell you what is wrong

SSkull995, 2019-06-01:
Hey! Yeah, I could have had, but I already had a look at it and nothing changes but the G29 presence in it. However, because anybody can make errors, I attach the two gcode files. The first one is without G29, while the second one is with the G29 command. I honestly don't understand why my TMCs became more sensitive, randomly. I only use Cura as a printing software

[CFFFP_xyz_sphere0.pdf](https://github.com/MarlinFirmware/Marlin/files/3243467/CFFFP_xyz_sphere0.pdf)
[CFFFP_xyz_sphere1.pdf](https://github.com/MarlinFirmware/Marlin/files/3243468/CFFFP_xyz_sphere1.pdf)

SSkull995, 2019-06-01:
Thank you all for the help in advance!

SSkull995, 2019-06-01:
I've just connected my printer to repetier, and I discovered a weird thing. When I do an x homing, and then a y homing, the x motor tends to push the head against the end of the rod. This happens even if I do a y homing before an x, the y motor push the head against the end of the y rod. I really don't know why this is happening, because the printer was totally fine just two days ago.

SSkull995, 2019-06-04:
No one can actually help me? Today a strange thing happened again, the y axis TMC was triggered during the senseless homing but not at y=0. I don't know why, but it seems that on the y axis the TMC finds an obstacle that is not there.

BastR, 2019-06-05:
DO you update your TMCSTEPPER library ?

BastR, 2019-06-05:
I had the same problem with version 0.3.3 or 0.3.4, with the last version (0.3.5) the problem with the X axis has completely disappeared.

SSkull995, 2019-06-05:
hi @BastR, I am looking at the library right now, but the version I can see is the 2.5.0

SSkull995, 2019-06-05:
No wait I found it, but I have never installed that library and Arduino never told me that it was missing during the compiling of the code

SSkull995, 2019-06-05:
ok something is very very weird. Now the distance between the actual rod end on the y axis and the home sensed by the TMC is increased consistently. Why is this happening?!

BastR, 2019-06-05:
I think you use the TMC2130 library, in marlin 2.0 there is no TMC2130 or 2208 library, there is an unified library called TMCstepper. Delete the old TMC2130 library and install the TMCstepper library: https://github.com/teemuatlut/TMCStepper

SSkull995, 2019-06-05:
Oh man! Thanks, will try right away

SSkull995, 2019-06-05:
Ah wait, I have the marlin 1.1.9
Should I install the 2.x?

SSkull995, 2019-06-05:
I just edited the configuration and configuration_adv of marlin 2.0.x but I still get the error from marlin_main.cpp that the TMCV2130 library is missing. There is no TMCStepper library used

BastR, 2019-06-05:
Oops, I thought you were on 2.0 because I had exactly the same problems as you when I switched to this version...
So in 1.1.9 you stay on the TMC2130 library, in 2.0 the TMCSTEPPER.

You can upgrade to version 2.0, I've been using it for 1 week and it's very stable

SSkull995, 2019-06-05:
Ok! I am trying to upgrade, but I am having many problems. Do you know where I could find a sort of tutorial on how to upgrade the firmware? Thanks

SSkull995, 2019-06-05:
I am trying to compile the code, I am getting some errors. One on bltouch, but I managed to solve it by enabling the pull-ups. The other one is relative to MAX6675_SS, which I have never used. I have _SS and not _SS_PIN, but I don't know where to change it. Did you encounter the same error?

SSkull995, 2019-06-05:
I forgot to update here, I apparently made it working! Thanks man!

SSkull995, 2019-06-09:
I noticed something weird, the problem keeps happening, but if I switch the printer off and wait like some hours, then it disappears

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle