MarlinFirmware/Marlin #14193: [FR]Help designing new Nextion display implementation

DotBreno, 2019-05-31:
Hi guys, I've been experimenting with Nextion display and Marlin for some time now, and have already tested the @apballard and @MagoKimbra versions and both works just fine. However since I learned about the Protocol Reparse mode (recmode=1) in Nextion, I belive it is possible to implement a display that communicate only via gcode trough serial port. Protocol Reparse mode lets you read and proccess raw serial data, also you can send raw serial data using 'print' function.

So far I managed to connect to an Arduino Mega2560 with Marlin installed, trough the Nextion simulation interface. I also changed the baudrate on Marlin to match the Nextion communication. When connected, Marlin sends the info message trough serial and I can read it on Nextion simulation interface, but when I try to send a gcode to the Arduino, the Tx led blinks but Marlin responds nothing. Summing up: I can read the first command Marlin sends via serial, but can't seem to send commands properly. I also tried sending "\r\n" in the end of each command, and disabling Protocol Reparse mode to see if only sending commands would work, but no luck.

[image]

Does anyone have any idea what should I do to get the Marlin to process the gcode commands I send trough Nextion? I would love to make this work and provide the project as open source after I finish.

MagoKimbra, 2019-05-31:
I use command printh.
Try printh 47 32 38 0D 0A
G28\x0D\x0A

DotBreno, 2019-05-31:
Thanks for the quick reply @MagoKimbra. And it worked like a charm c:. Guess I was missing the two ending bytes. 

Now I'm eager to get on with the work and share it when it's done. Thank you again.

closed 2019-05-31 by DotBreno