MarlinFirmware/Marlin #14194: [BUG] Bad Z Position after a power loss resume

andriee86, 2019-05-31:
### Description
After a power loss resume my printer rise up the nozzle too much making the function not working.
Skr 1.3 board, marlin 2.0.x release.

### Steps to Reproduce
1. Take off power during a print
2. Restart the print with the power resume menu

**Expected behavior:** resume the print from last know layer

**Actual behavior:** rising up the z axis too much

#### Additional Information

Video: https://www.youtube.com/watch?v=Z7xMlFywO8Q

yunstackCN, 2019-06-03:
I have the same problem.
Try the following changes, my problem is solved：
Modify file power_loss_recovery.cpp line 362, add following codes:
// Move back to the saved Z
  dtostrf(info.current_position[Z_AXIS], 1, 3, str_1);
  //sprintf_P(cmd, PSTR("G1 Z%s F200"), str_1);
  //gcode.process_subcommands_now(cmd);
    gcode.process_subcommands_now_P(PSTR("G1 Z0 F200")));
    
    // Restore Z position with G92.9
    dtostrf(info.current_position[Z_AXIS], 1, 3, str_1);
    sprintf_P(cmd, PSTR("G92.9 Z%s"), str_1);
    gcode.process_subcommands_now(cmd);

andriee86, 2019-06-04:
> I have the same problem.
> Try the following changes, my problem is solved：
> Modify file power_loss_recovery.cpp line 362, add following codes:
> // Move back to the saved Z [...]

mmmmm all the lines? some are "//"..... i tryed, it seemed working, i mean, the nozzle went back to right position but then rised up again....

yunstackCN, 2019-06-04:
"//" is code comment, not affects operatio.  raise how much？Is the z position value in lcd right ?

andriee86, 2019-06-04:
> 
> 
> "//" is code comment, not affects operatio. raise how much？Is the z position value in lcd right ?

yeah, but some of ur line code are commented, like: "//sprintf_P(cmd, PSTR("G1 Z%s F200"), str_1);
//gcode.process_subcommands_now(cmd);"
is it right or a mistake?

anyway i tryed a few times, with just the first part of ur code, the "// Move back to the saved Z" part, it does the first layer right but then on next layer is rising it up again,

and if u add all of ur code, with also the "// Restore Z position with G92.9" part, then it just go on the right layer but then rise it up again....

yunstackCN, 2019-06-04:
not mistake，it is just comment old code. please upload your  power_loss_recovery.cpp file，i check it

andriee86, 2019-06-04:
> 
> 
> not mistake，it is just comment old code. please upload your power_loss_recovery.cpp file，i check it

https://www.dropbox.com/s/xbb3578d6j3tb94/power_loss_recovery.cpp?dl=0

here we go, thanks mate :)

yunstackCN, 2019-06-05:
[G92.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/3257895/G92.cpp.zip)
[power_loss_recovery.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/3257899/power_loss_recovery.cpp.zip)
replace my files, have a try， need to decompress

tpruvot, 2019-06-05:
if you do a G1 Z0 onto the part to resume... will destroy it

andriee86, 2019-06-06:
> 
> 
> [G92.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/3257895/G92.cpp.zip)
> [power_loss_recovery.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/3257899/power_loss_recovery.cpp.zip)
> replace my files, have a try， need to decompress

thanks buddy!! its all working now with ur files! dont know what wrong, but its working so fair enough for me ^^ really thanks!

tpruvot, 2019-06-06:
yep, after a test.. my patch doesnt work, doesnt do a Z homing at all and seems to double the Z position

robbycandra, 2019-06-06:
Fixed with PR #14230

boelle, 2019-07-21:
@andriee86 can you confirm its fixed with PR #14230 ?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle