MarlinFirmware/Marlin #14199: LCD refresh Issue

augiem94, 2019-05-31:
I'm on 1.1.9 and using the miniviki display driver. When I open the menu from the info screen the top 8 pixels of the screen do not update until the encoder is moved. Any help would be appreciated, I can upload my configs when I get off work.

shitcreek, 2019-06-13:
You might just have a glitchy LCD. But as stated in the README:
Marlin 1.1.9 is the final release of the AVR-only flat version of Marlin Firmware, so there will be no further 1.1.x releases. However bugfix-1.1.x will continue to receive patches for critical bugs, **so be sure to test it (or bugfix-2.0.x) before reporting any bugs you find in 1.1.9.** 

You should try these other sources :
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2019-07-21:
@augiem94 did you try bugfix-2.0.x?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle