MarlinFirmware/Marlin #14202: [BUG] Wrong z-home with probe

guestisp, 2019-06-01:
Suddenly, the Z Home position is wrong. Maybe I did some bad configuration change (but I don't know which one, exaclty).

Usually , I had the Z-home position, after a G28, set to 10+z-probe-offset. So, with a z-probe-offset set to -2.50, my z home was 12.50

For a reason I don't know, now z home is shifter at about 5mm.
This is the log with z-offset set to 0. I would expect a z home set to 10, not 15

```
Send: M503
Recv:   G21 ; (mm)
Recv:   M149 C ; Units in Celsius
Recv:   M200 D1.75
Recv:   M200 D0
Recv:   M92 X80.00 Y80.00 Z400.00 E415.00
Recv:   M203 X300.00 Y300.00 Z5.00 E25.00
Recv:   M201 X1500.00 Y1500.00 Z100.00 E1500.00
Recv:   M204 P500.00 R500.00 T500.00
Recv:   M205 B20000.00 S0.00 T0.00 J0.08
Recv:   M206 X0.00 Y0.00 Z0.00
Recv:   M420 S0 Z0.00
Recv:   M145 S0 H210.00 B60.00 F0
Recv:   M145 S1 H250.00 B100.00 F0
Recv:   M301 P28.20 I2.71 D73.36
Recv:   M304 P183.88 I36.20 D233.48
Recv:   M413 S1
Recv:   M851 Z0.00
Recv:   M900 K0.00
Recv:   M603 L450.00 U450.00
Recv:   M425 F0.00 X0.00 Y0.00 Z0.00
Recv: ok P15 B3
Recv:  T:22.81 /0.00 (980.38) B:21.68 /0.00 (982.31) @:0 B@:0
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok P15 B3
Recv:  T:22.69 /0.00 (980.63) B:22.42 /0.00 (981.13) @:0 B@:0
Send: G28
Recv:  T:24.06 /0.00 (977.88) B:20.16 /0.00 (984.75) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.34 /0.00 (981.31) B:22.07 /0.00 (981.69) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:21.12 /0.00 (983.75) B:25.20 /0.00 (976.56) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:20.59 /0.00 (984.81) B:23.24 /0.00 (979.81) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.42 /0.00 (986.81) B:24.30 /0.00 (978.13) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.84 /0.00 (980.31) B:22.30 /0.00 (981.31) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 (980.69) B:22.58 /0.00 (980.88) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.75 /0.00 (980.50) B:20.43 /0.00 (984.31) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.09 /0.00 (979.81) B:22.77 /0.00 (980.56) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.56 /0.00 (978.88) B:23.01 /0.00 (980.19) @:0 B@:0
Recv: X:80.00 Y:125.00 Z:15.00 E:0.00 Count X:6400 Y:10000 Z:6000
Recv: ok P15 B3
Send: M114
Recv: X:80.00 Y:125.00 Z:15.00 E:0.00 Count X:6400 Y:10000 Z:6000
Recv: ok P15 B3
Recv:  T:22.91 /0.00 (980.19) B:20.12 /0.00 (984.81) @:0 B@:0
Recv:  T:20.19 /0.00 (985.63) B:25.31 /0.00 (976.31) @:0 B@:0
Recv:  T:20.97 /0.00 (984.06) B:23.16 /0.00 (979.94) @:0 B@:0
Recv:  T:22.03 /0.00 (981.94) B:22.38 /0.00 (981.19) @:0 B@:0
```

guestisp, 2019-06-01:
I'm at  fc52c43a

guestisp, 2019-06-01:
Tried even after upgrading to the latest commit. same result.
I'm almost sure to have something wrong in my config, any clue ?

reloxx13, 2019-06-01:
start with uploading ut config files

guestisp, 2019-06-01:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3244255/Configuration.zip)

Roxy-3D, 2019-06-01:
Did you do a M502 and M500 to update EEPROM settings after changing values?   If not...  probably the old values are still being used.

guestisp, 2019-06-01:
Did M502 as wrote above. No M500 and it's a test and i didn't reboot, no need to save the values

guestisp, 2019-06-03:
The attached photo is taken after moving Z axis to 0. In reality Z is still 5-6mm above the bed.
I have phisically more space on the axis, i can manually move the Z rod until It touch the bed thus the movement is not blocked in any way

[image]

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

guestisp, 2019-07-21:
Keep this open

I'm still facing this issue, actually i'm set at -3.20 mm as probe offset, It should be -2.50 because my nozzle is 2.50mm under the probe

AnHardt, 2019-07-21:
@guestisp 

What is the probes housing?
What is the probes trigger point?
How deep has the nozzle to dig into the bed until the probe will trigger?
Please reread the description of the probe-nozzle offsets with having in mind (probe = probes trigger point / not the lowest point of its housing).

closed 2019-10-12 by boelle