MarlinFirmware/Marlin #14203: [BUG]  1.1.9+bugfix FAST_PWM_FAN not working

FlailAway, 2019-06-01:
I need 5KHz, 5v-Peak-to-Peak to drive a laser. 
Marlin 1.1.9.
Anet A6
Anet board, current version.
I am using the input leg of the Fan MOSFET for the 5v but frequency rattles around near 980Hz.
I just installed the latest bugfix but still no change.
I found this here [Sound](https://github.com/MarlinFirmware/Marlin/issues/13571#issuecomment-483662065) but the Speaker is already disabled.
I tried using SOFT_PWM but setting SCALE to 1, gives 15KHz, too high.

I enabled FAST_PWM_FAN but it does not change frequency

### Steps to Reproduce
Enable FAST_PWM_FAN
Upload
M502
M500
M501
M106 S128

As above

**Expected behavior:** [What you expect to happen]
Frequency change

**Actual behavior:** [What actually happens]
No frequency change

FlailAway, 2019-06-06:
Bump? Has 1.1.9 been abandoned inf lieu of 2.0?

Anyone? Even the stock 7.xx-KHz shown in in the Wiki might be OK, but I cannot get that either. I need some help here guys.

boelle, 2019-07-21:
"Has 1.1.9 been abandoned inf lieu of 2.0?"

yes

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle