MarlinFirmware/Marlin #14207: [BUG] Error Compiling Marlin 2.0.x Bugfix for Use With BLTouch V3

AuntJemima, 2019-06-02:
### Description

Unfortunately, I've had zero luck getting my BLTouch V3 to work with my Ender 3, and have tried multiple configurations of both 1.1.x bugfix and TH3D firmware to no avail, so I'm trying to compile 2.0.x bugfix, but encountering an LCD error when enabling LCD_BED_LEVELING with BOARD_MELZI_CREALITY for my Ender 3.

### Steps to Reproduce
1) Download latest Marlin 2.0.x bugfix
2) #define MOTHERBOARD BOARD_MELZI_CREALITY in Configuration.h
3) #define LCD_BED_LEVELING in Configuration.h
4) (Optional) #define BABYSTEP_ZPROBE_GFX_OVERLAY in Configuration_adv.h
4) Attempt to compile

**Expected behavior:** Marlin successfully compiling

**Actual behavior:** Attempting to compile results in #error "LCD_BED_LEVELING requires a programmable LCD controller." (and #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD.")

#### Additional Information

ZIP file containing my `Configuration.h` and `Configuration_adv.h` files:
[Marlin_2.0.x_BLTouch_Config.zip](https://github.com/MarlinFirmware/Marlin/files/3244819/Marlin_2.0.x_BLTouch_Config.zip)

A bit of extra info about my BLTouch troubleshooting, in case it's helpful:
BLTouch enters error mode and z endstop triggers when I attempt to: deploy or stow the probe, and run the self-test. I can't convince it to do, well... pretty much anything.

### Full error output

```Arduino: 1.8.9 (Windows 10), Board: "Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

In file included from c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\marlinconfig.h:41:0,

                 from sketch\src\HAL\HAL_AVR\HAL_spi_AVR.cpp:38:

c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\SanityCheck.h:590:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."

     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."

      ^

c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\SanityCheck.h:1212:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."

     #error "LCD_BED_LEVELING requires a programmable LCD controller."

      ^

In file included from c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\marlinconfig.h:41:0,

                 from c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\gcode\gcode.h:275,

                 from sketch\src\gcode\geometry\G53-G59.cpp:23:

c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\SanityCheck.h:590:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."

     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."

      ^

c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\SanityCheck.h:1212:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."

     #error "LCD_BED_LEVELING requires a programmable LCD controller."

      ^

In file included from c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\marlinconfig.h:41:0,

                 from sketch\src\feature\bltouch.cpp:23:

c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\SanityCheck.h:590:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."

     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."

      ^
[... 250 lines cut ...]
      ^

In file included from c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\marlinconfig.h:41:0,

                 from c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\gcode\gcode.h:275,

                 from sketch\src\gcode\eeprom\M500-M504.cpp:23:

c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\SanityCheck.h:590:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."

     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."

      ^

c:\users\drago\appdata\local\temp\arduino_build_984588\sketch\src\inc\SanityCheck.h:1212:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."

     #error "LCD_BED_LEVELING requires a programmable LCD controller."

      ^

Error compiling for board Sanguino.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.```

sl1pkn07, 2019-06-02:
you set the correct LCD? in ender 3 with stock LCD is 

#define REPRAP_DISCOUNT_SMART_CONTROLLER

AuntJemima, 2019-06-02:
> 
> 
> you set the correct LCD? in ender 3 with stock LCD is
> 
> `#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`

Apparently I neglected to realize that there was a section in Configuration.h for LCD configuration! I found the one you mentioned, as well as:
//#define LCD_FOR_MELZI

Both of these give me an unsigned int error related to u8glib, and a cursory Google search reveals that my version of u8glib, 1.19.1, may be too recent, so I'm going to downgrade that and see what happens.

AuntJemima, 2019-06-02:
I was able to get it to compile using #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER after updating to Arduino 1.9.0 beta and u8g 1.19.1, thank you.

My BLTouch V3 still isn't behaving with 2.0.x, but at this point I'm 100% convinced that it's not a software/firmware issue, as I've spent countless hours now trying different configurations and always getting the same results.

closed 2019-06-02 by AuntJemima