MarlinFirmware/Marlin #14208: [BUG 2.0.x]  TMC2208 With TX pin only

nanokatz, 2019-06-03:
Quick question. I have been using bugfix 1.1.9 with my printer and wanted to try the 2.0.x dev branch.

I use TMC2208 I only have the TX lines connected at the moment. This works under 1.1.9 but it seems to me that the sanity checks in version 2 require both a TX and a RX pin to be given. 
Is it no longer possible to use the TMC2208 in this way?

LastDragon-ru, 2019-06-07:
You can remove sanity check and should work. But it may be a bit unstable with latest marlin and with new version of tmcstepper (0.3.4, 0.3.5) - sometimes one or two of three drivers on the same wire are not configured while boot.

nanokatz, 2019-06-07:
Okay good to know. I'll give it a shot. Each stepper has a separate tx line if that's what you mean by shared wires.

boelle, 2019-07-21:
is this a bug or question?

AmyTheCute, 2019-08-02:
> is this a bug or question?

It's a bug. this works fine in 1.1.9

boelle, 2019-09-24:
@nanokatz  is the problem still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

InsanityAutomation, 2019-10-15:
@boelle  This problem still exists however @teemuatlut has stated he may retire the feature. Personally I feel many want / need it but thats not my call.

boelle, 2019-10-15:
maybe @teemuatlut will reconsider it, if so we could reopen

but yeah not our call

teemuatlut, 2019-10-16:
The tx only feature will be retired if I get half-duplex UART working on all platforms.
If the comms would work with only one pin, it doesn't have enough benefits to do tx-only mode compared to the lost reading capabilities.

ManuelMcLure, 2019-10-16:
I'm willing to help with any testing to get half-duplex working on Re-ARM/RAMPS with BigtreeTech 2209s.

teemuatlut, 2019-10-16:
Thanks, appreciate it but Re-Arm (LPC) already supports half duplex and was written with that feature in mind =)
The bigger question is how AVR handles changing the pin mode just before reading.

ManuelMcLure, 2019-10-16:
Unfortunately, it's never worked for me. But that's probably more a hardware issue.