MarlinFirmware/Marlin #14209: [FR] Driver TMC 2209

biskero, 2019-06-03:
Hello,

any plans to add the TMC 2209 ?
Looks like it's a combination of the TMC 2130 and TMC 2208.

thx

teemuatlut, 2019-06-03:
Yes.

biskero, 2019-06-03:
very cool !

teemuatlut, 2019-06-03:
I should be receiving a set of drivers this week (maybe next) and I can then do some basic testing.
But I have very limited access to my printers this month and the support might first be at a theoretical level for a bit.

biskero, 2019-06-03:
sounds great, will order some for my skr v1.3 board

doggyfan, 2019-06-03:
They'll go nicely with the new board ;) .. which has a the lovely STM32F7 CPU :)

[image]

biskero, 2019-06-03:
wow!

codycato, 2019-06-03:
What's the difference between this and v1.3?

biskero, 2019-06-03:
from pic, better power connectors, 3 fuse, 6 drivers slots, a lot more pin connections.
But I have no idea for the rest.

stricktech, 2019-06-03:
awsome glad to hear the 2209s will get worked on.

Thats the skr pro, i should be getting one around the 25th of this month (won it in the btt FB group)
They announed it in the FB group but nor release date yet, and they havent said much about it, besides what you can see.

InsanityAutomation, 2019-06-03:
Yeah, id love to get one to throw an idex machine on here... Wonder why they want back to the STM32 chips though over the LPC, a bit more stable atm...

codycato, 2019-06-03:
I'll be o  the lookout, but I just got the v1.3. Still installing and configuring....if theres no major difference I'll just pass on this board

stricktech, 2019-06-03:
i read it was a stm32f4 , suppose to be more powerful. 
[image]

stricktech, 2019-06-03:
> I'll be on the lookout, but I just got the v1.3. Still installing and configuring....if theres no major difference I'll just pass on this board

i just got the skr 1.3 installed and set up on my cr10, the day before they announced i was one of the 10 they would be sending them to. depending on more info on the board, ill either tear cr10 back down, or use it on one of my other printers.

doggyfan, 2019-06-03:
The STM32F4/7 MCU's are **_much_** more powerful than the old LPC chips, they have hardware FPU, faster CPU clock, more pin options etc. Saying that, the old LPC MCU's do just fine anyway for a 3D printer.

InsanityAutomation, 2019-06-03:
> The STM32F4/7 MCU's are **_much_** more powerful than the old LPC chips, they have hardware FPU, faster CPU clock, more pin options etc. Saying that, the old LPC MCU's do just fine anyway for a 3D printer.

I'd say awesome in a couple months when we know the platform is stable! Before then questionable business decision but I won't complain too much as it gives us good accessible low cost hardware to debug on :)

bigtreetech, 2019-06-04:
please reference this [PR](https://github.com/MarlinFirmware/Marlin/pull/14215)

InsanityAutomation, 2019-06-04:
They found us talking about putting these on their boards! Thanks @bigtreetech im sure more of us will be getting these in our hands soon. I still cant find anywhere to buy a pro board yet :/

stricktech, 2019-06-04:
They also make tmc2209 and started selling them last week, so they need the marlin support, they have not started selling the skr pro yet, but they created a fb group last night for release Info (bigtreetech skr pro user group)

stricktech, 2019-06-04:
New image btt added to the group
[image]

InsanityAutomation, 2019-06-04:
Yeah, I hopped on the group last night as well. Ive got an idex machine here that could use a better board than the atmega2560....

Gerggg, 2019-06-11:
Hello, any news about this feature ?

Hedda, 2019-06-18:
FYI; both the FYSETC Cheetah board and BIQU BIGTREETECH SKR MINI-E3 boards are now Aliexpress

https://www.reddit.com/r/3Dprinting/comments/c1lttc/fysetc_cheetah_board_chinaclone_of_th3d_ezboard/

https://www.reddit.com/r/BIGTREETECH/comments/c0hdht/bigtreetech_skr_mini_e3_done/

Both these boards feature integrated TMC2209 drivers and STM32F103 32-bit based MCU

Gerggg, 2019-06-19:
Is the TMC 2209 compatible with Marlin's Linear advance?

LinoBarreca, 2019-06-27:
it has been merged already.
https://github.com/MarlinFirmware/Marlin/pull/14249

thisiskeithb, 2019-08-29:
@boelle: This issue can be closed since TMC2209s are supported.

boelle, 2019-08-30:
same as before, we can reopen if we get major protests :-D

closed 2019-08-30 by boelle