MarlinFirmware/Marlin #14213: Configuration Marlin FW 2.0.x dual Z MKS SBASE

TomTom076, 2019-06-03:
Hello!
Tried to configure dual Z in Marlin 2.0.x latest version, download today. I have done the config in configuration.h and configuration_adv.h and plattformio.ini.
Board => MKS SBASE V1.3
E1 stepper should be Z2
Anybody an idea whtas going wrong?
Stepper E1 isn´t moving while Z1 is moving. Z2 is wobbling and not moving.

Thanks!!!!

Tom

[Config MKS SBASE.zip](https://github.com/MarlinFirmware/Marlin/files/3248993/Config.MKS.SBASE.zip)

RudyBenner, 2019-06-06:
Corex machine? line 588
You did not define the stepper drivers. SBASE uses DRV8825. beginning line 644

Z2 is twitching? Wiring error for that stepper. Check the cable. 
Phases in series will produce low torque and ambiguous rotation.

TomTom076, 2019-06-07:
Hi!
Yes COREX. And there was also a wiring error! Z2 is moving.
Thanks!

closed 2019-06-07 by TomTom076