MarlinFirmware/Marlin #14217: G28

chenmaker, 2019-06-04:
i want turn on my 3Dprinter run a autohome(G28)
example-->https://github.com/Bougakov/Micromake-D1-3D-printer/blob/master/Installing%20custom%20firmware.md

shitcreek, 2019-06-12:
auto0.g is probably the closest thing Marlin provides - https://github.com/MarlinFirmware/Marlin/wiki/Autostart

thinkyhead, 2019-06-14:
We have a Micromake D1 and it can run Marlin. However, Marlin only comes with a C1 configuration. So it needs to be tweaked to run on the D1.

thinkyhead, 2019-06-14:
Here is the configuration we use with the Marlin `bugfix-1.1.x` branch (which is 1.1.9 plus a few tweaks). [Micromake D1.zip](https://github.com/MarlinFirmware/Marlin/files/3289250/Micromake.D1.zip)

Maybe it will work for you.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle