MarlinFirmware/Marlin #14219: Platform IO Compilation Error - LPC1768 with Arduino USB Host Shield (Max3421E)

msutas, 2019-06-04:
I receive errors related to multiple definition of structures for USB host shield. I don't know whether the problem is related with my setup or a bug, but it seems like a problem between the usb host shield code and the LPC1768 Arduino framework for platformio.

Board is BIQU SKR V1.1. Configuration files and the compiler output are attached.

@p3p maybe you may review the compiler output for addressing the issue.

[Marlin_usb_host_shield_compile_error.zip](https://github.com/MarlinFirmware/Marlin/files/3253836/Marlin_usb_host_shield_compile_error.zip)

p3p, 2019-06-04:
Unfortunately this looks like a deep routed incompatibility between the LPC176x Arduino frameworks usb stack and the Marlins Max3421E code, It seems no one has ever tried to use this on an LPC176x board.

Don't have time to look deeper right now but a fair few things would probably need renamed or removed from the global namespace to get this to build, though no idea if it would work correctly at that point without testing on this platform.

msutas, 2019-06-04:
OK since I know this is not a problem with my setup. I will try to eliminate the multiple definitions. If I can solve it, I will create a pull request.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle