MarlinFirmware/Marlin #14220: requesting information on fade height

leytanelectronics2, 2019-06-04:
I have been struggling to find any useful information on what the fade height is for my printer, I am using the anet a8 running on marlin 1.1.9 (i think) using the ramps 1.4 control board.

During the last few prints, I've done the first 1cm of the print comes out very irregular and lumpy on the outside like it's extruding more material than I programmed. 

While looking at the menu items I came across the fade height option which was set at 10, does this setting have anything to do with the first centimeter of my prints coming out over extruded or am I looking in the wrong place?

RudyBenner, 2019-06-06:
[image]

RudyBenner, 2019-06-06:
I would suggest setting fad height to zero to see what the resultant change is.
M420 S1 Z0

Vertabreak, 2020-01-14:
Fade tapers off the bed leveling over the set # in mm allowing the printers hardware to take over without mesh compensation at the set height.

Vertabreak, 2020-01-19:
@boelle i answered this should be safe to close.

closed 2020-01-24 by boelle