MarlinFirmware/Marlin #14221: [BUG 2.0.x ] (Last stepper.h halt my Delta)

dzieva, 2019-06-04:
### Description
My delta stopped moving properly, the steps got very slow and caught.

### Steps to Reproduce

Steeper.h has something wrong on line 168, but I also commented line 158 for prediction. now works as expected.

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

Please correct module stepper.h
[image]

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-07-19:
What exactly do you mean by 'with one more stepper'? Are all your drivers LV8792? or just one? Please clarify.

boelle, 2019-09-24:
@dzieva is this still a problem?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle