MarlinFirmware/Marlin #14224: [FR] Marlin 2.0.x TMC2208 always on Spreadcycle after powerup

rado79, 2019-06-05:
### Description

My new setup is a Bigtree SKR v1.3 with TMC2208 in UART mode from watterott. 
UART Connection is ok, stepper drivers are working. 
From Octoprint Terminal i can use gcode commands to change driver settings.
Switching between stealthchop and spreadcycle before and while printing with gcode M569 works fine.
But every powerup drivers are in spreadcycle mode.

### Steps to Reproduce

1. configure Marlin 2.0.x
2.  #define STEALTHCHOP_XY
3.  #define STEALTHCHOP_Z
4. compile and send to SD card with Atom
5. put SD card in SD Slot on SKR v1.3
6. power up
7. push reset button
8. M122 - stealthchop false
9. M569 S1 X Y Z.....
10. M122 - stealthchop true
11. powerdown - wait 10 seconds - power up
12. M122 - stealthchop false
#### Additional Information

Send: M122
Recv: 		X	Y	Z	Z2	E
Recv: Enabled		false	false	false	false	false
Recv: Set current	801	801	801	801	801
Recv: RMS current	795	795	795	795	795
Recv: MAX current	1121	1121	1121	1121	1121
Recv: Run current	25/31	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31	12/31
Recv: CS actual		12/31	12/31	12/31	12/31	12/31
Recv: PWM scale	26	29	26	14	14
Recv: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	false	false	false	false	false
Recv: msteps		16	16	16	16	16
Recv: tstep		max	max	max	max	max
Recv: pwm
Recv: threshold		98	100	658	658	263
Recv: [mm/s]		100.59	98.95	3.00	3.00	30.06
Recv: OT prewarn	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time		3	3	3	3	3
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv: -end		-1	-1	-1	-1	-1
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs
Recv: DRVSTATUS	X	Y	Z	Z2	E
Recv: stst		X	X	X	X	X
Recv: olb		X	X
Recv: ola		X	X
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0x80:0C:00:C0
Recv: 		Y	0x80:0C:00:C0
Recv: 		Z	0x80:0C:00:00
Recv: 		Z2	0x80:0C:00:00
Recv: 		E	0x80:0C:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
Recv: ok

gloomyandy, 2019-06-05:
May be a silly question but if you have eeprom enabled did you remember to reset to defaults and store the new settings in eeprom?

rado79, 2019-06-05:
Yes eeprom ist enabled. I changed Set current to 801mA in Marlin, it's up to date, same with thresold.

gloomyandy, 2019-06-08:
I think you need to post your configuration files. I have a similar setup (SKR V1.2 with TMC2208 drivers) and on that my TMC2208s are in stealthchop mode as selected. This was with today's Marlin build. Make sure you are using a recent version of Marlin (and not the bigtreetech version). As above be sure to do an M502 followed by M500 whenever you install a new build.

rado79, 2019-06-09:
Yes i always do a M502 M500 command. I downloaded the Marlin bugfix-2.0.x on 27.05.2019. How do i know the version number?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3270000/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3270001/Configuration_adv.zip)

btw: I compile with Atom/platformio. I added TMCSteppers 0.3.5 to libraries, but while compiling i think it uses 0.3.3

[image]
[image]

gloomyandy, 2019-06-10:
Please try again with the most recent version of Marlin and the most recent TMCStepper library there have been many changes since the version you are using!

rado79, 2019-06-11:
Update to the latest Marlin (10.06.2019) and TMCStepper 0.4.0 solved the problem.
Thank you.

closed 2019-06-11 by rado79