MarlinFirmware/Marlin #14226: [BUG] AKLP Delta Auto Calibration Probing Fail

OutspokenPlanet, 2019-06-05:
### Description

Delta Auto Calibration Probing Fail

### Steps to Reproduce

1. Home all (G28)
2. Start Auto Calibration (G33)

**Expected behavior:** Run through several iterations of bed probing

**Actual behavior:** Runs first iteration no issue, second iteration probe near centre of bed ok, then fails at approx 4mm height at the X tower and reports STOW PROBE, which then is followed by PROBING FAILED

I've tried this on the original Trigorilla board and also a BigTreeTech SKR 1.3, both produce the same outcome

[AKLP config files.zip](https://github.com/MarlinFirmware/Marlin/files/3257464/AKLP.config.files.zip)

closed 2019-06-11 by OutspokenPlanet

maxarchx, 2019-06-16:
Same issue, you fixed it?

raqun99, 2019-07-23:
Same issue, you fixed it?

pkoller, 2019-08-13:
Same issue, how to fix it?

OutspokenPlanet, 2019-08-26:
The fix seemed to be to make sure a decimal number is used in the delta section i.e. the delta height, delta radius etc.