MarlinFirmware/Marlin #14228: Rotary Delta(IcePick-Delta)

ur5zeb, 2019-06-06:
Can add kinematics delta robot
On the example of IcePick-Delta https://github.com/TTN-/IcePick-Delta

shitcreek, 2019-07-19:
please rephrase your question. Are you requesting this be added to Marlin?

ur5zeb, 2019-07-19:
Yes, I ask you to add this kinematics in Marlin 2

Ux1F595, 2020-05-01:
Hi there,
Any news on this FR? I am working on a custom rotary delta configuration as well,
and I would like to implement the rotary delta kinematics for Marlin.

As i am fairly new to Marlin any support would be highly appreciated.

For now i would start by adding the required **configuration**, as well as the forward/inverse kinematics in the **delta.h/delta.cpp** files.
The calculations and configuration parameters could be adopted from [https://www.marginallyclever.com/other/samples/fk-ik-test.html](https://www.marginallyclever.com/other/samples/fk-ik-test.html).

Are there any other modifications that need to be done in order to make the rotary delta work?

Im still not sure about the following points
- **speed/feedrate calculation**: As far as i have seen, the planner splits the movement into small steps and calculates the time this movement should take. This calculation is done in Cartesian coordinates, so the feedrate should still be valid if we introduce other kinematics. Am i right on this one or am i missing something?
- **Calculation time**: the forward/inverse kinematics are more complex than the trilateration used for deltas right now. Of course we should keep the equations as efficient as possible. As i am working on a Cortex-M7 board this should work on my system. But I won't be able to test performance for less powerful boards.
- Auto calibration features will not work on the new kinematics. I would have to test if this is needed .... And if so, see how it works and how we can make it work for rotary delta as well.

sjasonsmith, 2020-11-10:
@Ux1F595 are you still working on this?

I have considered this one of the least practical feature requests in here, because it is hard to find developers even to work on _normal_ delta issues, let alone completely unique kinematics. I had assumed there was no reason to keep this around, and only now noticed that you commented on it (albeit 6 months late!).

sjasonsmith, 2020-12-02:
Closing based on lack of interest. Adding motion systems adds complexity to Marlin which is only warranted if people are actually using them. Given the lack of response to my earlier post there seems to be little reason to keep this open.

closed 2020-12-02 by sjasonsmith