MarlinFirmware/Marlin #14231: [BUG] Printer suddenly sopps printing

davidhuwyler, 2019-06-06:
### Description
Sometimes the Printer stopps printing in the middle of a print. I print from SD Card.
The Hotend and the printbed stay hot and no error is on the display.

I tried printing tethered to the host pc (started the print in simplify 3d but still from sd card) to get some debug information.
As the printer stoped the console said: READ: echo:SD card ok

The printer stopps always at different positions in the print.
I tried different sd cards with no effect... 

### Setup info:
Bigtreetech SKR V1.3 Board with Marlin 2.0 from today.
Printing from SD card inside a Reprap discount graphics LCD.

gloomyandy, 2019-06-06:
My guess is that you have noise on the cables going to the SD card reader. You could try using the SD card reader on the skr board instead (you will need to change the config to do this). If you really want to use the LCD SD card reader you could try changing the speed used for the SD card spi bus.

davidhuwyler, 2019-06-06:
Im gonna try this thanks.

But I dont think its electromagnetic noise. i tried shorter cables far away from other cables... Ill report back

Grogyan, 2019-06-06:
You can try different QUALITY SD cards

davidhuwyler, 2019-06-06:
I used 2 different sandisk cards (2gb sd and 4gb sdhc) and 2 identical philips cards (16gb sdhc). i think they are not too bad quality cards...

davidhuwyler, 2019-06-07:
I just tried to use the onboard sd reader with the same result:
After a short while during printing, the printer stopped (with same output on the console `READ: echo:SD card ok`

I also tried to add MINIMUM_STEPPER_PULSE 2 in Configuration_adv.h which didnt have any effect either.

pritty frustrating... have no idea what to try next.

Powerprobot, 2019-06-07:
Stops the print always in the same height? I have problems with two Displays and i solved with an hardware fixup on the SKR 1.3 Board. The Powerline 5Volts are overseed with peaks/breaks. I stabilizied with an 100µF Elko in the 5Volts power. I have make the PDF to help other peoples with Display problems.  I think the distorted power treated the SD-Card readings.
Here is an PDF in German, i hope it is helping you: [SKR 5Volt Power](https://www.dropbox.com/s/7gfvd48doi537ce/Anleitung%20Bigtree%201.3%20mit%20Kondensator.pdf?dl=0)

davidhuwyler, 2019-06-08:
Thanks Powerprobot. Cool little mod. Im from switzerland so german is no problem for me :-)
I tried it out but it didnt help unfortunately.

gloomyandy, 2019-06-08:
So just to be clear, do you know when the message "echo:SD card ok" actually appears? Is it at the same time as the printer halts?

davidhuwyler, 2019-06-08:
Yes its exactly at that time. i saw it come up...

davidhuwyler, 2019-06-08:
Does anyone know if marlin stopps the print if a motor overheats with tmc2130 drivers? i just noticed that my extruder motor is really hot...

gloomyandy, 2019-06-08:
I don't think that Marlin has any way to detect an overheating stepper motor (unless it gets so hot that the motor coils short, which seems unlikely).

The SD card message is only displayed when the Sd card is initialized. As far as I can see this will only happen if either the Sd detect pin is triggered or something issues an M21 command. I assume you have not done anything like click on anything in your host software that might trigger an M21 command? I wonder if you are getting noise or false triggers on the SD card detect pin? One way to test that is to disable it. To do this edit the pins file for your board (pins_BIGTREE_SKR_V1.3.h) and right at the end of the file add the line...
```C
#undef SD_DETECT_PIN
```
If this works then if you boot Marlin without the SD card inserted and then insert the card it should not be detected and you will need to to use the InitSD card menu item to enable it. If that all seems ok then try another test print.

davidhuwyler, 2019-06-08:
I tried printing with the internal sd card which dosnt even have a deteced pin (you have to init it with M21) but the same error occoured.

My guess is still it has something to do with the hot motor. I reduced the current to 0.5A (titan aero works fine with this low current setting) and finished a 3h print. Im nor sure jet but i think this fixed the issue. No idea why dough...

gloomyandy, 2019-06-08:
The internal SD card reader will only disable the SD detect pin if you also enable sharing of the SD card via USB, I wasn't sure if you has done that or not. If you don't do this then the SD detect pin is still defined (even though it is for the wrong SD card reader!). Did you enable the USB sharing feature when you tested using the internal reader (by enabling USB_SD_ONBOARD in the advanced config)?

This is the code...
```C
#elif ENABLED(LPC_SD_ONBOARD)

  #if ENABLED(USB_SD_ONBOARD)
    // When sharing the SD card with a PC we want the menu options to
    // mount/unmount the card and refresh it. So we disable card detect.
    #define SHARED_SD_CARD
    #undef SD_DETECT_PIN
    //#define SD_DETECT_PIN  P0_27   // (57) open-drain
  #endif

  #define SCK_PIN          P0_07
  #define MISO_PIN         P0_08
  #define MOSI_PIN         P0_09
  #define SS_PIN           P0_06   // Chip select for SD card used by Marlin
  #define ONBOARD_SD_CS    P0_06   // Chip select for "System" SD card

#endif
```

davidhuwyler, 2019-06-08:
Ah ok nice to know. Yes I enabled that.

davidhuwyler, 2019-06-09:
Reducing the current through the extruder solved my problem. was no marlin bug but nontheless very strange... so im gonna close the issue

closed 2019-06-09 by davidhuwyler

davidhuwyler, 2019-09-15:
it depends on your stepper drivers. if you controll the drivers with spi/uart you can limit the current in the firmware otherwise you can turn the pot on the driver with a screwdriver

Rugsejis, 2019-09-23:
Me need help, i have printer with skr v1.3 motherboard and used tmc 2130 standlone drivers.printer suddenly stoppd printing, i printing from sd card (insert to BIGTREETECH TFT35). But when printing from usb cable, printer dont stops. When i print from sd card not for all files printer stopps sometimes print more then 20 hours and dont stops. It is possible print files from motherboard sd card reader?

Cemileri, 2019-11-30:
I was wondering if this situation occurs because of vibration when using a sd card with adapter..