MarlinFirmware/Marlin #14232: [Bugfix 2.0.x] BIQU SKR 1.3 TMC2130 Dual Z - Only one moves

Rexima, 2019-06-06:
Hi everybody,

i have an AM8, i mean, everything is replaced except the steppers and the rods 👯‍♂ .
And i replaced my MKS Gen 1.4 Board with the BIQU SKR 1.3 and i switched from TMC2208 to TMC2130.
SPI is working, i mean i guess:
`Send: M122
Recv: 		X	Y	Z	Z2
Recv: Enabled		false	false	false	false
Recv: Set current	760	760	760	760
Recv: RMS current	734	734	734	734
Recv: MAX current	1035	1035	1035	1035
Recv: Run current	23/31	23/31	23/31	23/31
Recv: Hold current	11/31	11/31	11/31	11/31
Recv: CS actual		11/31	11/31	11/31	11/31
Recv: PWM scale	35	34	0	0
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	false	false
Recv: msteps		16	16	16	16
Recv: tstep		max	max	max	max
Recv: pwm
Recv: threshold		52	52	658	658
Recv: [mm/s]		152	152	3	3
Recv: OT prewarn	false	false	false	false
Recv: off time		4	4	4	4
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		1	1	1	1
Recv: Stallguard thrs	8	8	0	0
Recv: DRVSTATUS	X	Y	Z	Z2
Recv: stallguard		X	X	X
Recv: sg_result		0	0	267	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb					X
Recv: ola					X
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 		X	0x81:0B:00:00
Recv: 		Y	0x81:0B:00:00
Recv: 		Z	0x81:0B:01:0B
Recv: 		Z2	0xE0:0B:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: ok`

It stays on enabled=false, and i dont know why.
I have also an problem with the second Z2 Stepper, its not moving.

On my MKS Gen 1.4 everything works with both Z stepper drivers and steppers, so i dont know, why its not here working...

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3262121/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3262122/Configuration_adv.txt)

shitcreek, 2019-06-13:
Sorry to hear, but glad you've found the issue. Please kindly close the issue. Cheers.

boelle, 2019-07-21:
closing as OP found the issue was his hardware

closed 2019-07-21 by boelle