MarlinFirmware/Marlin #14234: [FR] Add hotend heating condition to start controller fan

Patag, 2019-06-06:
Hi all,
Because of my very noisy E3DV6  heatsink fan, I wanted to switch it off unless an ongoing print.
So I wired it with my controller box fan. To trigger it, I modified some Marlin code in **controllerfan.cpp**. It seems to work fine, at least for a single extruder configuration.
My post goal is to propose it as a PR but I'm a real noob with Github/PR management. So let me share my code modifications here :

`// If any of the drivers or the hotend or the bed are enabled...
    if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON 
      #if HAS_HEATED_BED
        || thermalManager.temp_bed.soft_pwm_amount > 0
      #endif
      #if HOTENDS
        || thermalManager.temp_hotend[0].soft_pwm_amount > 0
        #if HOTENDS > 1
          || thermalManager.temp_hotend[1].soft_pwm_amount > 0
          #if HOTENDS > 2
            || thermalManager.temp_hotend[2].soft_pwm_amount > 0
            #if HOTENDS > 3
              || thermalManager.temp_hotend[3].soft_pwm_amount > 0
              #if HOTENDS > 4
                || thermalManager.temp_hotend[4].soft_pwm_amount > 0
                #if HOTENDS > 5
                  || thermalManager.temp_hotend[5].soft_pwm_amount > 0
                #endif // HOTENDS > 5
              #endif // HOTENDS > 4
            #endif // HOTENDS > 3
          #endif // HOTENDS > 2
        #endif // HOTENDS > 1
      #endif // HOTENDS`

Bye

PS : Marlin/Github tutorial welcome

gloomyandy, 2019-06-06:
Why not just use the existing feature that only turns the fan on when the hotend is above a certain temperature?

Patag, 2019-06-07:
Hi @gloomyandy, thanks for your reply. What is this feature ?
Or maybe I haven't been clear enough in the description : I'm talking about the fan fitted on the extruder heat-sink (normally always on) not the one used to cool-down the newly extruded filament on the bed.
If this feature already exists, I'll use it (don't want to re-invent the wheel).
Thanks

kAdonis, 2019-06-07:
@Patag Look at Configuration_adv.h line 348, for SKR1.3 try:
```cpp
#define E0_AUTO_FAN_PIN P2_04
 ```

Patag, 2019-06-07:
@kAdonis Thanks for your reply. Anyway, I wanted to keep the "Controller Fan" feature (triggered on stepper spinning or bed heating) which is already linked to P2_04 pin. SanityCheck rejects this additional pin assignment (a very good thing in fact). Moreover, my real board fan is also wire on this pin.
So never mind, I'll will keep on living with my local code modifications.

closed 2019-06-07 by Patag