MarlinFirmware/Marlin #14237: [BUG] Marlin crashes with default Cura settings

ReProgram8a9ce45a, 2019-06-07:
### Description
Marlin crashes when encountering certain g-code created by Cura 4.1

By default, Cura sets a unlimited speed amount for the Z axis, producing this G code 
;TYPE:SUPPORT
G1 F17987547480000 Z0.3

When Marlin encounters this it freezes and then the watch dog resets

### Steps to Reproduce
Cura 4.1 with defaults settings for Z speed

#### Additional Information
Marlin 2.0, downloaded on May 24 2019, MKS board with atmel 2560

2kingsrock, 2019-06-07:
I opened a bug with the CURA team over this, as I ran into this several weeks ago.  I was told that this default value was used by the devs as it is the speed of light.  I told them as funny as this is to use as a new default, it is a nonsense value that will cause problems with many machines.  The issue is the system is set with a default value of 0 in the software.  The callout box says it will default to light speed if a non value is set, or it will use the default in the firmware.  However, the system will hang when it parses the G1 F17987547480000 line (basically at any Z hop command) when trying to compare that to the firmware values stored in marlin.  The easy fix is to put a value in like 25mm or something reasonable in the slicer.  I am not sure it is the same, but is dividing by zero a bug? :)  I still think the DEVs for CURA need to change this default to anything, even 99999 would be great 🥇   https://github.com/Ultimaker/Cura/issues/5753

Roxy-3D, 2019-06-07:
It may be reasonable to re-open #13929 for discussion.    In particular...  I think vendors have a legitimate need to set various limits for their machines and what has been tested and verified on their platforms.

More in depth thinking here:  https://github.com/MarlinFirmware/Marlin/pull/13929#issuecomment-490505972

InsanityAutomation, 2019-06-11:
I'll drop in support here, even though it's redundant with my PR mentioned. I'd like to clamp jerk, Accel, junction deviation, and feedrate to sane limits.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle