MarlinFirmware/Marlin #14238: [BUG] Build failure SKR 1.3

xxsashixx, 2019-06-08:
Compile error with SKR 1.3 on with Visual Studio Code

Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
In file included from C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:33:0,
from Marlin\src\HAL\HAL_LPC1768\../../core/../HAL/HAL_LPC1768/HAL.h:40,
from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:25,
from Marlin\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp:26:
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/const_functions.h:65:120: warning: 'nodiscard' attribute directive ignored [-Wattributes]
[[nodiscard]] constexpr uint32_t map(uint32_t x, uint32_t in_min, uint32_t in_max, uint32_t out_min, uint32_t out_max) noexcept{
^~~~~~~~
In file included from C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:33:0,
from Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/HAL_LPC1768/HAL.h:40,
from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:30,
from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:23:
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/const_functions.h:65:120: warning: 'nodiscard' attribute directive ignored [-Wattributes]
[[nodiscard]] constexpr uint32_t map(uint32_t x, uint32_t in_min, uint32_t in_max, uint32_t out_min, uint32_t out_max) noexcept{
^~~~~~~~
In file included from C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:33:0,
from Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/HAL_LPC1768/HAL.h:40,
from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:30,
from Marlin\src\HAL\HAL_LPC1768\HAL_timers.cpp:31:
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/const_functions.h:65:120: warning: 'nodiscard' attribute directive ignored [-Wattributes]
[[nodiscard]] constexpr uint32_t map(uint32_t x, uint32_t in_min, uint32_t in_max, uint32_t out_min, uint32_t out_max) noexcept{
^~~~~~~~
In file included from C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:33:0,
from Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/HAL_LPC1768/HAL.h:40,
from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:30,
from Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:51:
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/const_functions.h:65:120: warning: 'nodiscard' attribute directive ignored [-Wattributes]
[[nodiscard]] constexpr uint32_t map(uint32_t x, uint32_t in_min, uint32_t in_max, uint32_t out_min, uint32_t out_max) noexcept{
^~~~~~~~
In file included from C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/gpio.h:5:0,
from C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:35,
from Marlin\src\HAL\HAL_LPC1768\../../core/../HAL/HAL_LPC1768/HAL.h:40,
from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:25,
from Marlin\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp:26:
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/registers.h:8:64: warning: 'nodiscard' attribute directive ignored [-Wattributes]
[[nodiscard]] constexpr auto memory_ptr(const std::size_t loc) noexcept {
^~~~~~~~
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/registers.h:13:65: warning: 'nodiscard' attribute directive ignored [-Wattributes]
[[nodiscard]] constexpr auto& memory_ref(const std::size_t loc) noexcept {
^~~~~~~~
In file included from C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/gpio.h:6:0,
from C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:35,
from Marlin\src\HAL\HAL_LPC1768\../../core/../HAL/HAL_LPC1768/HAL.h:40,
from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:25,
from Marlin\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp:26:
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/bit_manipulation.h:8:55: warning: 'nodiscard' attribute directive ignored [-Wattributes]
[[nodiscard]] constexpr auto bit_value(const Bit bit) noexcept {
^~~~~~~~
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/bit_manipulation.h:13:72: warning: 'nodiscard' attribute directive ignored [-Wattributes]
[[nodiscard]] constexpr bool bit_test(const Value& val, const Bit bit) noexcept {
^~~~~~~~
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/bit_manipulation.h:18:57: warning: 'nodiscard' attribute directive ignored [-Wattributes]
[[nodiscard]] constexpr auto bitset_value(Args... args) noexcept {
^~~~~~~~
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/bit_manipulation.h:23:80: warning: 'nodiscard' attribute directive ignored [-Wattributes]
[[nodiscard]] constexpr auto bitset_mask(const Value val, const BitSet bitset) noexcept {
^~~~~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\MarlinSerial.h:24:0,
from Marlin\src\HAL\HAL_LPC1768\MarlinSerial.cpp:26:
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h: In member function 'void HardwareSerial<RXB_SIZE, TXB_SIZE>::begin(uint32_t)':
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h:161:8: error: expected '(' before 'constexpr'
[... 241 lines cut ...]
from C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:34,
from Marlin\src\HAL\HAL_LPC1768\main.cpp:10:
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h: In member function 'void HardwareSerial<RXB_SIZE, TXB_SIZE>::begin(uint32_t)':
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h:161:8: error: expected '(' before 'constexpr'
if constexpr (TXB_SIZE > 0) {
^~~~~~~~~
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h: In member function 'size_t HardwareSerial<RXB_SIZE, TXB_SIZE>::write(uint8_t)':
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h:204:8: error: expected '(' before 'constexpr'
if constexpr (TXB_SIZE > 0) {
^~~~~~~~~
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h:238:3: error: expected '}' at end of input
}
^
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h:238:3: warning: no return statement in function returning non-void [-Wreturn-type]
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h: In member function 'void HardwareSerial<RXB_SIZE, TXB_SIZE>::flushTX()':
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h:241:8: error: expected '(' before 'constexpr'
if constexpr (TXB_SIZE > 0) {
^~~~~~~~~
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h: In member function 'void HardwareSerial<RXB_SIZE, TXB_SIZE>::IRQHandler()':
C:\users\sashi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h:310:8: error: expected '(' before 'constexpr'
if constexpr (TXB_SIZE > 0) {
^~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o] Error 1
=============================================================================================== [ERROR] Took 2.84 seconds ===============================================================================================

gloomyandy, 2019-06-08:
I've just built Marlin for the SKR V1.3 using the very latest source for everything and in VSCode, all works fine for me. Make sure you have everything up to date (especially platformio) and try it again. If you are still having problems then post your configuration files.

KBurdin, 2019-06-24:
build it on 64 bit machine

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sjasonsmith, 2019-09-21:
A lot of us build for SKR 1.3 regularly. I did it a few moments ago with no issue. Given the lack of any additional information I think this should be closed.

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

butchs38, 2020-01-09:
I am also having a problem if I install vs code on another machine copy marlin folder over to it it will build no errors. On my main pc I uninstalled vs code reinstalled and tried to build the same folder. I just keeping errors. I dont think its anything in config.h or adv.h since it builds on another pc. Here are the errors

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1768; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: framework-arduino-lpc176x 0.2.2, toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)    
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...    
Dependency Graph
|-- <Servo> 1.0.0        
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1   
|-- <TMCStepper> 0.6.1   
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\timers.cpp.o
In file included from c:\users\butch\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\string:52,
                 from c:\users\butch\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\stdexcept:39,
                 from c:\users\butch\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\array:39,
                 from C:\users\butch\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pin_control.h:3,
                 from C:\users\butch\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:20,
                 from C:\users\butch\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from v:\marlin-2.0.1\marlin\src\hal\shared\marduino.h:36,
                 from v:\marlin-2.0.1\marlin\src\hal\hal_lpc1768\hal.h:39,
                 from v:\marlin-2.0.1\marlin\src\hal\hal.h:26,
                 from v:\marlin-2.0.1\marlin\src\inc\marlinconfig.h:30,
                 from v:\marlin-2.0.1\marlin\src\core\serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
c:\users\butch\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\bits\basic_string.h:39:10: fatal error: ext/atomicity.h: No such file or directory
 #include <ext/atomicity.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from c:\users\butch\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\string:52,
                 from c:\users\butch\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\stdexcept:39,
                 from c:\users\butch\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\array:39,
                 from C:\users\butch\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pin_control.h:3,
                 from C:\users\butch\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:20,
                 from C:\users\butch\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from v:\marlin-2.0.1\marlin\src\hal\shared\marduino.h:36,
                 from v:\marlin-2.0.1\marlin\src\hal\hal_lpc1768\hal.h:39,
                 from v:\marlin-2.0.1\marlin\src\hal\hal.h:26,
                 from v:\marlin-2.0.1\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
c:\users\butch\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\bits\basic_string.h:39:10: fatal error: ext/atomicity.h: No such file or directory
 #include <ext/atomicity.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o] Error 1
In file included from c:\users\butch\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\string:52,
                 from c:\users\butch\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\stdexcept:39,
[... 119 lines cut ...]
[... cut ...]
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
rumba32_f446ve                IGNORED
mks_rumba32                   IGNORED
include_tree                  IGNORED
=============================================================================== 1 failed, 0 succeeded in 00:00:32.347 ===============================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.