MarlinFirmware/Marlin #14240: [FR] Anet A8 PLUS support

XOIIO, 2019-06-08:
Hi there, I was wondering if this has been tested on the A8 PLUS as of yet? I just finished assembling mine, though it does have a couple quirks to it, not sure if it's due to assembly as it's my first 3d printer, or if it's due to the firmware.

The print bed size has been increased, but I'm not sure if there are any revisions to the main board at all, or differences between the old and new control panel, I'd be happy to provide pictures if needed to compare/

govido, 2019-06-10:
The board looks nearly the same apart from the beefed up screw connectors, the screen is the same as the Anet A6. So a config for the A6 should get you started with needs for tweaking bowden settings and bed size.

XOIIO, 2019-06-12:
I gave it a shot with the stock config, and it seemed to be working well however when moving my printer the bed slid, and it seems to have taken out one of the stepper drivers and killed the board, even soldered a new atmega1284p but the furthest I can get is flashing the bootloader, and that axis' stepper driver heats up even while disconnected.

The only problem I had that stopped be from testing further aside from this is that I need to set my x axis offset around 24mm from the home switch, as it's got quite the offset from the edge of the bed to the switch, so I was unable to level the bed properly to give the printer a real test before the board died.

When trying to set the offset however I got "err too far", and looking into it, apparently you can only set an offset 20mm or so? I'm not sure how I can fix this issue, there's no way to assemble the printer so
 that the bed lines up with the end stop from what I can see.

jonasborneland, 2019-08-06:
I'm running mine with marlin :) – https://github.com/jonasborneland/anet-a8-marlin-config-file

closed 2020-01-03 by XOIIO