MarlinFirmware/Marlin #14241: [BUG] POWER SUPPLY

Amachete, 2019-06-08:
Hello 

I do not know whats wrong but the lates versions i have found problems with the firmware.

POWER SUPPLY =0 Compiles and work
POWERT SUPPLY =1 or 2 compiles but It gets frozen and the LCD does not show anything

I have a misc of TMC and Allegros.

Let me know if you need something

I usually work with POWER SUPPLY=2

Thank you all!!!

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3268450/Marlin.zip)

Amachete, 2019-06-08:
I have reading the code a bit, and the configuration files.

If you have 
POWER_SUPPLY=2

and disable

//#define AUTO_POWER_CONTROL        // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS           // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
   #define AUTO_POWER_CHAMBER_FAN
    #define POWER_TIMEOUT 30
  #endif
 
I works and run on MEGA

This feature worked for me a month ago, more or less...

I would like to help you more, but my code knowledge is limited....

Let me know if you need something else...

Amachete, 2019-06-10:
I have just tried the version 1.1.9 and I have the same issue...
Power=1 and 2 compiles but gets stuck at boot
Power=0 compiled and works

I do not know what's wrong with my set up...how can find out if my mega-ramps works fine?

The blink sketch works fine,...something else to try?

Thank you

Grogyan, 2019-06-10:
Are you able to try marlin 2.0 and see if the problem is still there?

Amachete, 2019-06-11:
I will try this evening I will try again, with a fresh directory of files.

I will post my results.

Thank you for your reply.

Amachete, 2019-06-11:
Hello again.

I have configured the configuration files again from a clean Bugfix 2.0.

My Set up with this 2 options enabled does not work:

#define POWER_SUPPLY 2
#define AUTO_POWER_CONTROL 

The LCD display is white, without anything on it, any letter, nothing. It looks locked in a bootloop.

On the other hand.

With this options:
#define POWER_SUPPLY 2
//#define AUTO_POWER_CONTROL 

With these options the screen looks fine. but I can start up my machine, due to source power has swich managed by the RAMPS-MEGA.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3278599/Marlin.zip)

Here you have my latest setup.

Let me know if you(someone) find out something else.....

I will try next weekend the bugfix 1.1.9,If I find some free time.

Thank you.

Amachete, 2019-06-14:
Hello again

I have been doing more trys.......

If I disable the TMCdrivrers with power control ON

#define AUTO_POWER_CONTROL

and disabled the TMC drivers

//#define X_DRIVER_TYPE  TMC2130
//#define Y_DRIVER_TYPE  TMC2130
//#define Z_DRIVER_TYPE  TMC2130
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE TMC2130
//#define Z3_DRIVER_TYPE A4988
//#define E0_DRIVER_TYPE TMC2130
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988

This way the LCD works and shows information.

Other detail to mention, RAMPS is working on USB, the printer is unplugged. When is plugged the power source starts to work but the LCD shows nothing and I never shuts off as it used to.

¿Is this a bug?

Can someone help me? thank you!!

Amachete, 2019-08-02:
I have just updated with the latest MARLIN2.0 02/08/2019

The bug is still there.

When I activate  #define AUTO_POWER_CONTROL

Marlins compiles and uploads...but the screen goes white!!

the configuration files keeps the same, tmc2130, capacitive sensor, and smart reprap lcd, xbox style power source nothing fancy,  

let me know if you need more information!!!!

shitcreek, 2019-08-17:
First question: did it ever work for you before?
In the latest Marlin, for the xbox power supply you need to set
PSU_ACTIVE_HIGH to true, and wire the blue wire from the PSU to the 5v on the board (be sure to wire the ground as well).

Amachete, 2019-08-20:
Hello 

Thank you for yor replay,

I can Swicth ON and OFF the power supply with the GCODE M80 and M81. It woks without any problem.

What does not work is the AUTO_POWER_CONTROL.

[Wiring diagram ](https://reprap.org/wiki/File:Wiring_scheme.png)

The PSU is On at start up, in order not to have problem with my TMC2130, because these drivers need to have the main voltage on at the boot up.

Let me know if you need any other piece of information.

Thank you

shitcreek, 2019-08-25:
Have you tried compiling with only one of the dependants inside of `#if ENABLED(AUTO_POWER_CONTROL)` at a time to see if you could narrow down the culprit?

Amachete, 2019-09-08:
Sorry for my delay answering you!!

I will try narrow down this thing, I attach you my configuration files, My marlin is the last version, I use git in order to have it updated.

//define AUTO_POWER_CONTROL
compiles and uploads and works right

  #define AUTO_POWER_CONTROL  // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS           // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
    #define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 30
  #endif
  
 Compiles, uploads but DO NOT WORKS on MEGA,etc
 
   #define AUTO_POWER_CONTROL  // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS           // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
    //#define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 30
  #endif
 
 DISABLING AUTO_POWER_CHAMBER OPTION
 
 Compiles, uploads but DO NOT WORKS on MEGA,etc 
 
 I Compiled with AUTO_POWER_CONTROL disabled again just in case there is something wrong with my MEGA, and IT Ccompiled, upoaled and worked as suppoused.(When I say IT WORKS, MARLIN SHOWS the main menu after boot up)
 
   #define AUTO_POWER_CONTROL  // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS           // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    //#define AUTO_POWER_CONTROLLERFAN
    //#define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 30
  #endif
  
  DISABLING AUTO_POWER_CONTROLLERFAN
 
 Compiles, uploads but DO NOT WORKS on MEGA,etc 
 
   #define AUTO_POWER_CONTROL  // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS           // Turn on PSU if fans need power
    //#define AUTO_POWER_E_FANS
    //#define AUTO_POWER_CONTROLLERFAN
    //#define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 30
  #endif 
 
  Compiles, uploads but DO NOT WORKS on MEGA,etc 
  
  
     #define AUTO_POWER_CONTROL  // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    //#define AUTO_POWER_FANS           // Turn on PSU if fans need power
    //#define AUTO_POWER_E_FANS
    //#define AUTO_POWER_CONTROLLERFAN
    //#define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 30
  #endif 
 
 Compiles, uploads but DO NOT WORKS on MEGA,etc 
 
 
 I do not know whatelse I can do, If you need any other test, I will try...Let me know.
 
 Thnak you.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3587327/Marlin.zip)

boelle, 2019-10-12:
@Amachete still having issues?

Amachete, 2019-10-13:
> @Amachete still having issues?

Yes, I have  just updated with the latest source, and I still have issues!!

If I activate the #define AUTO_POWER_CONTROL the source complies, uploads to the mega but It does not boot up! the screen is white, any letter symbol, nothing.....

If I deactivate I works fine, At least It boots up.

//#define AUTO_POWER_CONTROL

Anyone else having the same issue?

Let me know If you have any other piece of information.

Bye

boelle, 2019-10-24:
how does the crash manifest?
are the TMC's powered up correctly?

Amachete, 2019-10-25:
Yes they are working gine

When I disable that option it works fine and tmc2130 work fine, the power
source is on from the beginning.
When I enable that option....
If I upload the new firmware and it does nothing, the LCD is on without
anything on it. The power source is on....I have checked the a marlin a
week old. I will update this weekend.

Let me know if I can tell you something else, It's quite weird...

Bye

Amachete, 2019-10-31:
I have just updated with the latest Marlin....
And I am afraid It keeps on crashing on Booting up.....when It is activated #define AUTO_POWER_CONTROL
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3795742/Marlin.zip)

I do not know how I can help you out to solve the problem....

boelle, 2019-11-08:
@shitcreek does this sound right?

shitcreek, 2019-11-08:
I tested an ATmega2560 board (Creality v2 to be exact) with a selection of different stepper drivers defined:
Works regardless of `AUTO_POWER_CONTROL`: A4988, DRV8825, LV8729, TB6560, TMC2130_STANDALONE, TMC2208
Doesn't work with `AUTO_POWER_CONTROL` : TMC2100, TMC2130, TMC2160

I'm not certain of the difference between TMC2130_STANDALONE and TMC2130, but would it work for you in the meantime?

shitcreek, 2019-11-09:
The hack suggests in https://github.com/MarlinFirmware/Marlin/issues/14558 works.
Comment/disable `restore_stepper_drivers();` in `power.cpp`

MrAlvin, 2019-11-10:
Perhaps this gives you some hardware clues https://learn.watterott.com/silentstepstick/faq/#boards-with-usb-power-supply

And perhaps these will as well (give hardware clues) https://learn.watterott.com/silentstepstick/faq/#silentstepsticks-with-variable-3-5v-logic-voltage

https://learn.watterott.com/silentstepstick/faq/#what-is-the-difference-between-silentstepsticks-with-3-5v-and-5v-logic-voltage

A hardware fix could perhaps be to add a shottkey diode from VIO to VM  (or better from a 5V power source to VM. 

Otherwise I think that the software solution is mentioned in the comment by shitcreek (just above this comment).

Amachete, 2019-11-11:
Hello there

I must say that I already have installed a protector thing under the
tmc2130, the one that watterott designed in order to avoid such a problem.

I will try that hack someone(sorry I am writing with my phone) wrote before.

I will tell something else, when I tried.

In my opinion the problem is with inicilization of SPI...or something.

Thanks

boelle, 2019-11-21:
@Amachete still an issue?

boelle, 2019-11-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-11-24 by boelle