MarlinFirmware/Marlin #14242: [BUG] TMC2130 not moving

pauhull, 2019-06-08:
I recently bought some BigTreeTech TMC2130 drivers. I intended to use them in SPI mode, but they where pre-configured, so I had to use them in standalone mode. They didn't seem to work and they had some other issues so I decided to buy Fysetc TMC2130 V1.2 drivers which come in SPI mode by default. I installed them and wired the SPI connection (see picture below). They have the same issue as the old BigTreeTech drivers, instead of moving the stepper motors just stall, altough the new drivers don't make such a loud sound.

describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Patag, 2019-06-08:
Sounds like stepper cable issue. Depending on the steppers, cables may differ. I experienced this kind of issue.

pauhull, 2019-06-08:
The wiring of the steppers is 2B 2A 1A 1B, just like with the A4988s

Patag, 2019-06-08:
Check (via wires colors) if it comply with stepper/board connections.

pauhull, 2019-06-08:
I still have the proble,

pauhull, 2019-06-09:
In some tutorials I saw people soldering those little solder pads on the underside together like this: https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcQH4VugYm4p5k7LBaZm4tEJnJ-Ng0Rp3OTxX0e6pGQ2WpS-F59LxA

The left solder pad came disconnected from factory, but the little ones on the right aren't soldered together at all. I'm not sure whether to connect them at all because I'm afraid to destroy something with my bad soldering skills and I'm not even sure if this is causing the problem. I read that this is only neccessary to enable StallGuard2 (which I don't need).

kAdonis, 2019-06-09:
Your SPI connection is working so no need to solder. Your M122 shows ola and olb, which means: open load detected. This could be a cable issue or a false warning, but in your configuration,h you have:
```cpp
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 634, 638, 3170, 765 } 
```
these values are very high, are you sure you used 1/16 microsteps before?
if the video in the first post is from your printer, I would try this values
```cpp
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 765 }
```

pauhull, 2019-06-09:
I changed my default steps/mm, but that didn't change anything. The stalling only was a bit quieter (you almost can't hear it anymore). I think the problem is the detected open load. This is really weird since the printer has worked perfectly before and the only thing I did was swapping the A4899 drivers and changing the options in the firmware.

pauhull, 2019-06-10:
I think I solved the problem. I soldered these little bridges (CFG4 to GND, CFG5 to VIO) on the y axis, and after that the y axis started working like intended. However the z axis also started working, but the x axis didnt. I soldered the bridges on the x axis and then it started working, too. The crosses at ola and olb partly disappeared. Thank you for your help!

closed 2019-06-10 by pauhull