MarlinFirmware/Marlin #14244: [BUG] Delta G33 X and Y PROBE_OFFSET_FROM_EXTRUDER

mignolo4, 2019-06-08:
Not completely sure if this is a BUG or not (I've read other closed issues without understanding it completely), but when I start the delta auto configuration from the LCD it doesn't take into account the X Y PROBE_OFFSET_FROM_EXTRUDER configured in the configuration.h section;

my parameters:
`#define X_PROBE_OFFSET_FROM_EXTRUDER -32   // X offset: -left  +right  [of the nozzle]`
`#define Y_PROBE_OFFSET_FROM_EXTRUDER 40    // Y offset: -front +behind [the nozzle]`
`#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.25  // Z offset: -below +above  [the nozzle]`

This leads to a series of problems like probing outside the bed or to a not precise levelling.

No problem at all if I send, for instance, a `M48 P4 X0 Y0 V2 E1 L2`  or a `G30 X0 Y0`: the probe (BLtouch) is in the correct position (at the center of the plate X0,Y0);

but if I send a G33 command, the probing starting point 0,0 is reached by the nozzle and not by the probe (the LCD says X0 Y0 too);
when the automatic procedure probe a point close to the edge of the bed, the BLTouch sensor is obviously outside the fisical bed area with the consequences you can imagine.

Tested with the last 2.0 bugfix version: today 8 June 2019.

shitcreek, 2019-06-09:
@LVD-AC would be best person address this as they wrote it.

LVD-AC, 2019-06-09:
For calibration the nozzle needs to be in the center (all carriages at the same distance) and the nozzle needs to be at points in circle around that center. And to solve the probe trying to probe outside the bed radius you need to set DELTA_CALIBRATION_RADIUS in Config.h or with M665 Bxx.x so it doesn't.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle