MarlinFirmware/Marlin #14248: Use four endstop

nosleduc, 2019-06-08:
I have a project that needs four endstops on four different axes.
The endstop x_min, y_min and z_min are used as usual. I would like to use x_max as a min endstop for the fourth axis.

How can I change the code to do that ?

I'm using the version 2.0 of Marlin.

shitcreek, 2019-06-09:
This isn't something Marlin can do out the box, and would require a bit of customization which might be better labeled a Feature Request.

That said: this Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

DerAndere1, 2019-06-10:
This is also a duplicate of existing issues. There is this [recent Feature request 13813](https://github.com/MarlinFirmware/Marlin/issues/13813) and [the Old discussion in issue 8519](https://github.com/MarlinFirmware/Marlin/issues/8519) with Solutions.

boelle, 2019-07-21:
closing, as @shitcreek said this is not place to ask questions, and its also a dublicate

closed 2019-07-21 by boelle