MarlinFirmware/Marlin #14253: since updating MCStepper V0.4.0 I no longer have access to my serial port

gauix4, 2019-06-09:
SKR 1.3 TMC5160s V1.0
since updating MCStepper V0.4.0 I no longer have access to my serial port

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3269833/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3269834/Configuration_adv.txt)

gloomyandy, 2019-06-10:
It seems this issue is being worked on here https://github.com/MarlinFirmware/Marlin/pull/14245

teemuatlut, 2019-06-11:
I applied your configuration to fix_tmc5160_init (which I had rebased to bf-2.0.x).
Then I disabled any TMC drivers and updated the firmware on the SD card.
The board still froze.

I then checked out the unmodified branch and enabled a basic 4xTMC5160 based configuration with a few options. The board booted just fine.

For now I don't think your issue is with the TMC library.

I'll do some further digging making changes to your files and see what comes of it.
I'd suggest you re-do configuration and carefully **avoid** setting the max speeds to 900000.

teemuatlut, 2019-06-11:
[Marlin issue 14253.zip](https://github.com/MarlinFirmware/Marlin/files/3278280/Marlin.issue.14253.zip)
Boot working config based on the provided files.

boelle, 2019-07-21:
so this one is fixed?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

gauix4, 2019-07-23:
the problem has never been solved
this one still does not work
I simply find how to counter the problem but not solve it

gauix4, 2019-07-23:
I just use the port
#define SERIAL_PORT 2 - 1
and it works
  but not the port
#define SERIAL_PORT_1 -1
it does not work

boelle, 2019-09-24:
@gauix4 is the issue still there?

closed 2019-10-12 by boelle