MarlinFirmware/Marlin #14254: SKR 1.3 TMC2130 SPI Dual axis grinding noise

MichaelCereda, 2019-06-09:
I have a CR-10S and few days ago I decided to change the logic board with an SKR 1.3 and TMC2130.

X and Y axis work pretty well and they are silent.
The dual Z axis is giving me headaches.
I put the 2 stepper motors in series following https://www.instructables.com/id/Wiring-Your-Z-Stepper-Motors-in-Series/ and the motors work pretty well when not connected to the screws. 
Once I connect them to the machine they make a grinding noise and suddenly jump forward to make the step.

Here's my current configuration
https://github.com/MichaelCereda/Marlin/pull/1 

also I've noticed parameters changing before/after trying to move the motor.

After a movement I can also hear a high pitch noise coming from the motors that lasts for a while after they move.

```
SENT: M122
READ: 		X	Y	Z	E
READ: Enabled		false	false	false	false
READ: Set current	800	800	800	800
READ: RMS current	795	795	795	795
READ: MAX current	1121	1121	1121	1121
READ: Run current	25/31	25/31	25/31	25/31
READ: Hold current	12/31	12/31	12/31	12/31
READ: CS actual		12/31	12/31	12/31	12/31
READ: PWM scale	0	0	0	0
READ: vsense		1=.18	1=.18	1=.18	1=.18
READ: stealthChop	true	true	false	true
READ: msteps		16	16	16	16
READ: tstep		max	max	max	max
READ: pwm
READ: threshold					
READ: [mm/s]					
READ: OT prewarn	false	false	false	false
READ: OT prewarn has
READ: been triggered	false	false	false	false
READ: off time		3	3	3	3
READ: blank time	24	24	24	24
READ: hysteresis
READ: -end		-1	-1	-1	-1
READ: -start		1	1	1	1
READ: Stallguard thrs	0	0	0	0
READ: DRVSTATUS	X	Y	Z	E
READ: stallguard					
READ: sg_result		0	0	0	0
READ: fsactive					
READ: stst		X	X	X	X
READ: olb					
READ: ola					
READ: s2gb					
READ: s2ga					
READ: otpw					
READ: ot					
READ: Driver registers:
READ: 		X	0x80:0C:00:00
READ: 		Y	0x80:0C:00:00
READ: 		Z	0x80:0C:00:00
READ: 		E	0x80:0C:00:00
READ: 
READ: 
READ: Testing X connection... OK
READ: Testing Y connection... OK
READ: Testing Z connection... OK
READ: Testing E connection... OK```

after moving 1mm (and hearing the terrible grinding noise)

```SENT: M122
READ: 		X	Y	Z	E
READ: Enabled		true	false	true	false
READ: Set current	800	800	800	800
READ: RMS current	795	795	795	795
READ: MAX current	1121	1121	1121	1121
READ: Run current	25/31	25/31	25/31	25/31
READ: Hold current	12/31	12/31	12/31	12/31
SENT: M105
READ: CS actual		12/31	12/31	12/31	12/31
READ: PWM scale	53	0	0	0
READ: vsense		1=.18	1=.18	1=.18	1=.18
READ: stealthChop	true	true	false	true
READ: msteps		16	16	16	16
READ: tstep		max	max	max	max
READ: pwm
READ: threshold					
READ: [mm/s]					
READ: OT prewarn	false	false	false	false
READ: OT prewarn has
READ: been triggered	false	false	false	false
READ: off time		3	3	3	3
READ: blank time	24	24	24	24
READ: hysteresis
READ: -end		-1	-1	-1	-1
READ: -start		1	1	1	1
READ: Stallguard thrs	0	0	0	0
READ: DRVSTATUS	X	Y	Z	E
READ: stallguard				X	
READ: sg_result		0	0	0	0
READ: fsactive					
READ: stst		X	X	X	X
READ: olb					
READ: ola				X	
READ: s2gb					
READ: s2ga					
READ: otpw					
READ: ot					
READ: Driver registers:
READ: 		X	0x80:0C:00:00
READ: 		Y	0x80:0C:00:00
READ: 		Z	0xA1:0C:00:00
READ: 		E	0x80:0C:00:00
READ: 
READ: 
READ: Testing X connection... OK
READ: Testing Y connection... OK
READ: Testing Z connection... OK
READ: Testing E connection... OK```

Can I provide other useful information? Thank you

shitcreek, 2019-06-10:
The current for your Z motors probably need to be higher as they are moving a large mass against gravity and using lead screws. Adjust the physical VREF (plenty of how-to's on YT) and/or M907 - http://marlinfw.org/docs/gcode/M907.html

That said: this Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

MichaelCereda, 2019-06-10:
Thank you. I'll close the issue and post in the right location.

closed 2019-06-10 by MichaelCereda