MarlinFirmware/Marlin #14255: [BUG] Axis Max out with Print Pause

CAP1Sup, 2019-06-10:
My Y axis maxed out after parking. This is due to my home position on the Y axis being 5, not 0. The park position is set to 0, 0 because I forgot to change it. An error needs to be added so that park positions may not be outside of the printing area.

ManuelMcLure, 2019-06-10:
Usually if your Y endstop is to the left of the bed, you set ``Y_MIN_POS`` to a negative number so your home position (front left corner of the bed) is 0,0. This makes your printer coordinates match your slicer coordinates and avoids a lot of problems down the road, such as this one.

CAP1Sup, 2019-06-10:
Yeah, the problem is that the nozzle sits 5 mm in on the bed when it hits the endstop. Therefore, in order to prevent future occurrences, it would be nice to have an error in place.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle