MarlinFirmware/Marlin #14256: Mega overclocking

3dmecanico, 2019-06-10:
This days i was working in a new hardware MKS GEN L 1.0 . I changed de speed from 16 to 20 MHZ. It's totally stable. I compiled for 20Mhz  marlin 1.1.9 and it's ok. I ask: What 's the  advantage  of  this in the planner secction. I see that at 20MHZ is a table for steppers. The max frecuency of the stepper rate up 25%?. It's multipulse begin at the same frecuency rate.
Generally ,this 25% of increment of speed is practical visible?
Thanks

boelle, 2019-07-21:
We're not a support group

This Issue Queue is for Marlin bug reports and development-related
issues, and we prefer not to handle user-support questions here. (As
noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/
1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-ijust-have-a-question).) For best results getting help with configuration
and troubleshooting, please use the following resources:

-RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
-[Tom's 3D Forums](https://discuss.toms3d.org/)
-Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
-Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
-[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a
bug in Marlin, then you should post a [bug report](https://github.com/
MarlinFirmware/Marlin/issues/new/choose).

boelle, 2019-10-24:
better ask here:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

this list is for bugs and fr's only

closed 2019-10-24 by boelle