MarlinFirmware/Marlin #14257: Marlin 1.1.9 thermocouple(AD8495) Weird calculation?

guana19, 2019-06-10:
when i tried to use thermocouple in on my printer with ad8495 breakout board, it would only measure something like 180C at room temp

oringinal marlin code: 
((RAW) * 6.6 * 100.0 / 1024.0 / (OVERSAMPLENR) * (TEMP_SENSOR_AD595_GAIN) + TEMP_SENSOR_AD595_OFFSET)

modified code:
((((RAW) * 5.0 / 1024.0 / (OVERSAMPLENR) - 1.25) / 0.005 / (TEMP_SENSOR_AD8495_GAIN)) + TEMP_SENSOR_AD8495_OFFSET)

the equation to measure temps with ad8495 is [Temperature = (Vout - 1.25) / 0.005 V ]

and i can't understand where did [(RAW) * 6.6 * 100.0 / 1024.0] came from!

as far as i know, oversamplenr means marlin will measure the temperature 16times and use the average of it.
so RAW input would be something like X16 of analog reading, which would be why we use RAW/OVERSAMPLENR.

since analog input is read as (analog read voltage/5)*1024,

i did this [(RAW) * 5.0 / 1024.0 / (OVERSAMPLENR)] to regain original voltage reading from the breakout board and applied it to ad8495 equation.

And now, i'm getting accurate temperatures.
i'm not sure if this is the right way
since there are now subtract in the code, i'm not sure if it will be ok to use
why the oringinal code was like that? where did that *6.6 came from?

AnHardt, 2019-06-12:
Different ADC reference voltages.

guana19, 2019-06-13:
> Different ADC reference voltages.
yeah i see that 6.6*100 = 3.3/0.005
but i still don't get it.
why would there be a need to make a false equation which cannot be used without modifying. (which needs an understanding of what the arduino input looks like, and oversampling) when we have solid equation about specific board![Temperature = (Vout - 1.25) / 0.005 V ]
yes i see that i could just change 6.6 to 10.0 and put a -1.25*200 as TEMP_SENSOR_AD8495_OFFSET
but why don't make it a default and make a option or remark for ref voltage

AnHardt, 2019-06-13:
Let's restart.
> when i tried to use thermocouple in on my printer with ad8495 breakout board, it would only measure something like 180C at room temp

You have an AD8495 module and get much to high readings in Marlin 1.1.9 .

> oringinal marlin code:
((RAW) * 6.6 * 100.0 / 1024.0 / (OVERSAMPLENR) * (TEMP_SENSOR_AD595_GAIN) + TEMP_SENSOR_AD595_OFFSET)

That's not Marlin code. It's locking like ([1.1.9](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/temperature.cpp#L892), [bf-1.1.9](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/temperature.cpp#L892) or [bf/2.0.x](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp#L1186)). You mixed up the formulas for AD595 and AD8495.

> modified code:
((((RAW) * 5.0 / 1024.0 / (OVERSAMPLENR) - 1.25) / 0.005 / (TEMP_SENSOR_AD8495_GAIN)) + TEMP_SENSOR_AD8495_OFFSET)

> the equation to measure temps with ad8495 is [Temperature = (Vout - 1.25) / 0.005 V ]

Ups. That's locking uncommon. Would you please give us a **link** to **your** AD8495 module.
As far as i know, at the [cheap modules](https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20190613015508&SearchText=ad8495) **REF** is connected to **GND** (by default), what results in being not able to measure negative temperatures (0° = 0V) (we don't need that) with a normal CPUs ADC. Your special module could have **REF** at 1.25V to make that possible (0° = 1.25V). 

> and i can't understand where did [(RAW) * 6.6 * 100.0 / 1024.0] came from!

When we began to support thermocoupler amplifiers we started with the AD595 and all the processors we supported ran on 5V. AD595 are not usable for 3.3V systems without making some handstands ([mainly because of min. 5V differenc votage](https://www.analog.com/media/en/technical-documentation/data-sheets/AD594_595.pdf)). So someone added the [AD8495](https://www.analog.com/media/en/technical-documentation/data-sheets/ad8494_8495_8496_8497.pdf) for his 3.3V system. 
Now since AD595 is out of production, its [price](https://de.aliexpress.com/wholesale?catId=0&initiative_id=SB_20190613030608&SearchText=ad595) is increasing month by month and modules for it seem to have [vanished](https://de.aliexpress.com/wholesale?catId=0&initiative_id=AS_20190613030609&SearchText=ad595+module), the pressure on Marlin increases to support AD8495 on 5V systems. That's perfectly possible by using the correct ADC reference voltage in the formular, but ([last time we discussed that](https://github.com/MarlinFirmware/Marlin/issues/11658)) we could not find a way to detect that automatically. So the problem remained unsolved. Do you have a good idea for that?

However.
> ((((RAW) * 5.0 / 1024.0 / (OVERSAMPLENR) - 1.25) / 0.005 / (TEMP_SENSOR_AD8495_GAIN)) + TEMP_SENSOR_AD8495_OFFSET)

Seems not to be the best idea, especially not the -1.25 at that place.
As long the build in formula has the form `RAW * WHATEVER_CONSTANT_FACTOR * TEMP_SENSOR_AD8495_GAIN` + TEMP_SENSOR_AD8495_OFFSET, any linear output of the amplifier can be adjusted to match the real °C reading just by adjusting TEMP_SENSOR_AD8495_GAIN and TEMP_SENSOR_AD8495_OFFSET, which are configurable. The numbers just may not be intuitive.

AnHardt, 2019-06-13:
Try to set TEMP_SENSOR_AD8495_GAIN to 3.3/5.0 or 5.0/3.3.
Compile. Than take a look at the resulting temperature at room temperature.
Subtract that value from room temperature and put the number in TEMP_SENSOR_AD8495_OFFSET.

AnHardt, 2019-06-13:
Looks as if i'm wrong. All of the [cheap modules](https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20190613065944&SearchText=ad8495) seem to have **REF** at 1.24 or 1.25 V now. You just have to look at the right offers. :-(
[Search for the circuit diagram](https://www.aliexpress.com/item/32963596648.html?spm=2114.search0104.3.9.d39048a5ZTLrYy&ws_ab_test=searchweb0_0%2Csearchweb201602_1_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103%2Csearchweb201603_51%2CppcSwitch_0&algo_expid=a1cf2c3a-3a9c-4cf8-b577-ac8b6f56e3f2-1&algo_pvid=a1cf2c3a-3a9c-4cf8-b577-ac8b6f56e3f2)
[Look at the backside](https://www.aliexpress.com/item/32875890159.html?spm=2114.search0104.3.143.d39048a5ZTLrYy&ws_ab_test=searchweb0_0%2Csearchweb201602_1_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103%2Csearchweb201603_51%2CppcSwitch_0&algo_expid=a1cf2c3a-3a9c-4cf8-b577-ac8b6f56e3f2-21&algo_pvid=a1cf2c3a-3a9c-4cf8-b577-ac8b6f56e3f2)
 
Not that nice. That limits the 3.3V systems to about 400°C.

masibu-labs, 2019-06-21:
I came across the same problem today while retro fitting an old Makerbot replicator. The AD8495 is one of the cheap models with a REF of 1.24V and a  transfer function (I think this is the right term?) of  5 mV / ° C. I am running it on a 5V system. 

Why not define the sensor reference voltage and transfer function along with the Gain and offset and update the formula to reflect this?  

Configuration_adv.h
`#define TEMP_SENSOR_AD8495_REFERENCE_VOLTAGE     1.24`
`#define TEMP_SENSOR_AD8495_TRANSFER_FUNCTION     0.005`

temperature.cpp
`#define TEMP_AD8495(RAW) ((RAW) * 5.0 / 1024 / (OVERSAMPLENR) - (TEMP_SENSOR_AD8495_REFERENCE_VOLTAGE)) / (TEMP_SENSOR_AD8495_TRANSFER_FUNCTION) * (TEMP_SENSOR_AD8495_GAIN) + (TEMP_SENSOR_AD8495_OFFSET)`

Ive tested this as standalone on an Arduino uno and was getting an output +2C greater than ambient temperature, however when added to marlin to test I was getting +9C than ambient. I'm not entirely sure how the RAW is determined. I was using analogRead() on my stand alone test and was just sampling single reads.

AnHardt, 2019-06-22:
RAW is just OVERSAMPLENR 10 bit ADC readings added.
The most common error, when getting far off readings, is connecting the AD8495 to a normal temperature input (with a 4.7K pullup resistor). It needs an input without a pullup resistor.

Of cause measuring temperatures with a AD8495 depends much on the ADC reference voltage. An absolute voltage is compared to the reference voltage. When measuring thermistors we compare resistentes where the pullup resistor is pulled to the reference voltage - reference voltage cancels out of the equation - for that we get the tolerances of the pullup resistor as an additional source of error.

boelle, 2019-07-21:
We're not a support group

This Issue Queue is for Marlin bug reports and development-related
issues, and we prefer not to handle user-support questions here. (As
noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/
1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-ijust-have-a-question).) For best results getting help with configuration
and troubleshooting, please use the following resources:

-RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
-[Tom's 3D Forums](https://discuss.toms3d.org/)
-Facebook Group ["Marlin Firmware"](https://www.facebook.com/
groups/1049718498464482/)
-Facebook Group ["Marlin Firmware for 3D Printers"](https://
www.facebook.com/groups/3Dtechtalk/)
-[Marlin Configuration](https://www.youtube.com/results?
search_query=marlin+configuration) on YouTube
-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a
bug in Marlin, then you should post a [bug report](https://github.com/
MarlinFirmware/Marlin/issues/new/choose).

boelle, 2020-06-20:
@guana19 is the issue gone?

boelle, 2020-06-23:
Anything before 2.0.0 is not being worked on anymore 

Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the issue isn't resolved yet, create a new bug report

closed 2020-06-23 by boelle