MarlinFirmware/Marlin #14258: [FR] RPW-Ultra thermistor table

VanessaE, 2019-06-10:
This is not actually a feature request.

Because I use an older firmware version, I am not confident that a proper fork/Pull Request would work out, so this is basically a "Please merge the following changes, I know they work" request.   As code goes, this change is trivial.

The RPW-Ultra hotend by Reifsnyder Precision Works comes with an unknown brand of 100k/NTC thermistor, due to an upstream vendor error.  As one might imagine, none of Marlin's existing tables work for it, though a couple come somewhat close.

So, since the thermistor works and is predictable, it doesn't make sense to replace it.  Since the manufacturer has a lot of them on-hand, it is likely that future RPW-Ultras will come with the same one.  Therefore, I've made a new table from scratch to match it.

Calibration methodology (warning!  bad science ahead! :grin:):

* Using a PeakMeter PM6501 with included thermocouple probe
* RPW-Ultra hotend, installed as part of a normal, production setup, with thermistor screwed into the upper thermistor hole; heat block is not insulated
* thermocouple probe inserted into the Allen socket hole of the grub screw in the hotend's secondary thermistor hole (I didn't have a suitable wrench to remove it)
* thermocouple oriented to get the hottest possible reading
* meter placed on my desk, propped upright, a few feet away from the hotend, with the thermocouple wire feeding through a slit in the printer's enclosure
* desk fan positioned to blow across it to prevent local heat build-up
* printer enclosure doors removed.
* Recorded ADC readings for 30° through 270° in 10° steps
  * Commanded printer to heat up to to somewhere close to the desired temperature, by guessing
  * tweak set temperature up/down a little at a time, until the printer's reading stops oscillating, and  thermocouple remains to within about ±0.2° of the desired temperature (well, until I got above 199.9°, beyond which my meter reads only in 1° increments).
* calculated resistances from the ADC readings for 30°, 150°, and 250° using `R = 4700 * ADC / (1024 - ADC)`
* plugged those resistance/temperature pairs into a Steinhart-Hart table generator
* double-checked the generated table against my recorded measurements
* flashed the firmware with the new table added and selected, and checked a few temperatures to make sure the printer's readings match the thermocouple after sufficient settling time.

The table below is the generated one, since it is a very close match to my recorded values.

For what it's worth, I chose 502 here because RPW-Ultra used to be known as the genuine J-head 12, but was re-branded, and it was originally supposed to use thermistor type 5 like previous J-head hotend models, if not for the above-mentioned upstream vendor error.

Marlin/src/module/thermistor/thermistor_502.h :

```c++
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

// 100k thermistor supplied with RPW-Ultra hotend, 4.7k pullup

const short temptable_502[][2] PROGMEM = {
  {OV(26),  300 },
  {OV(28),  295 },
  {OV(30),  290 },
  {OV(32),  285 },
  {OV(34),  280 },
  {OV(37),  275 },
  {OV(39),  270 },
  {OV(42),  265 },
  {OV(46),  260 },
  {OV(49),  255 },
  {OV(53),  250 }, // 256.5
  {OV(57),  245 },
  {OV(62),  240 },
  {OV(67),  235 },
  {OV(73),  230 },
  {OV(79),  225 },
  {OV(86),  220 },
  {OV(94),  215 },
  {OV(103), 210 },
  {OV(112), 205 },
  {OV(123), 200 },
  {OV(135), 195 },
  {OV(148), 190 },
  {OV(162), 185 },
  {OV(178), 180 },
  {OV(195), 175 },
  {OV(215), 170 },
  {OV(235), 165 },
  {OV(258), 160 },
  {OV(283), 155 },
  {OV(310), 150 }, // 2040.6
  {OV(338), 145 },
  {OV(369), 140 },
  {OV(401), 135 },
  {OV(435), 130 },
  {OV(470), 125 },
  {OV(505), 120 },
  {OV(542), 115 },
  {OV(579), 110 },
  {OV(615), 105 },
  {OV(651), 100 },
  {OV(686),  95 },
  {OV(720),  90 },
  {OV(751),  85 },
  {OV(781),  80 },
  {OV(809),  75 },
  {OV(835),  70 },
  {OV(858),  65 },
  {OV(880),  60 },
  {OV(899),  55 },
  {OV(915),  50 },
  {OV(930),  45 },
  {OV(944),  40 },
  {OV(955),  35 },
  {OV(965),  30 }, // 78279.3
  {OV(974),  25 },
  {OV(981),  20 },
  {OV(988),  15 },
  {OV(993),  10 },
  {OV(998),   5 },
  {OV(1002),  0 },
};
```
Save the above with the indicated file name

Add the relevant few lines in `thermistors.h`:
```diff
  #include "thermistor_501.h"
#endif
+ #if ANY_THERMISTOR_IS(502) // 100k thermistor in RPW-Ultra hotend, Pull-up = 4.7 kOhm, "unknown model"
+  #include "thermistor_502.h"
+ #endif
#if ANY_THERMISTOR_IS(6) // beta25 = 4092 K, R25 = 100 kOhm, Pull-up = 8.2 kOhm, "EPCOS ?"
```
...and add an appropriate description to `Configuration.h`:
```diff
 *   501 : 100K Zonestar (Tronxy X3A) Thermistor
+ *   502 : 100k thermistor supplied with RPW-Ultra hotends (4.7k pullup)
 *     6 : 100k EPCOS - Not as accurate as table 1 (created using a fluke thermocouple) (4.7k pullup)
```

AnHardt, 2019-06-10:
Nice. Methodology seems to be correct.
The table could be a tiny bit better if the ADCvalues would have 2 or 3 decimals. (The OV(x) values are multiplied by 16 and become integers thereafter.)

VanessaE, 2019-06-10:
For whatever reason, the generator (a spreadsheet I grabbed from Thingiverse) only used integers.  Or maybe I did something wrong.  Still seemed to work okay. :-)

AnHardt, 2019-06-10:
How about https://www.thingiverse.com/thing:3300383 ?

VanessaE, 2019-06-10:
That one just gives an empty page when I tell it to generate a table.  I guess something about it isn't LibreOffice compatible.  Just to make sure I did it right:  
[image]
(note: it says "R (ohm)" on cell `I1`, but the after-loading presets in the spaces below it are clearly kOhms, so that's what I used, just as with [the original spreadsheet](https://www.thingiverse.com/thing:103668).)

InsanityAutomation, 2019-06-10:
I'll have to try it in a recent libreoffice version and see what happens. Iirc when I updated that spreadsheet for marlin 2.0 formatting I used Excel 2016.

VanessaE, 2019-07-02:
As an aside, I think this should be renumbered to 512 instead of 502 (think "should be type 5, on a J-head 12").

VanessaE, 2019-07-05:
@InsanityAutomation  did you ever fix that spreadsheet?

VanessaE, 2019-07-05:
I guess further discussion should be directed to the pull request.

closed 2019-07-05 by VanessaE