MarlinFirmware/Marlin #14259: [FR] LCD games should respect reversed direction input devices

pinchies, 2019-06-10:
e.g. Invaders game: multiply encoderPosition by (1-2*ui.encoderDirection)

    // Update Cannon Position
    int16_t ep = int16_t(ui.encoderPosition);
    int16_t ep = int16_t((1-2*ui.encoderDirection)*ui.encoderPosition);
    ep = constrain(ep, 0, (LCD_PIXEL_WIDTH - (CANNON_W)) / (CANNON_VEL));
    ui.encoderPosition = ep;
    ui.encoderPosition = (1-2*ui.encoderDirection)*ep;

https://github.com/pinchies/Marlin/commit/bccb2ac72e26085787dedc7d2c9a9903668c6a82#diff-4f26cb56f12a074a344c2db8bf3b5fae

shitcreek, 2019-06-10:
@thinkyhead

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

pinchies, 2019-07-21:
I posted the fix but I don’t know if it has been integrated? @thinkyhead

boelle, 2019-07-21:
did you submit a PR?

pinchies, 2019-07-21:
No... simply because I thought maybe my code was probably clunky and there would be a simple fix. The diff was supposed to show the way I did it, but I was confident that maybe somebody would use it to make a more efficient method.

boelle, 2019-07-21:
but i think thinkyhead will pick up on this soon

pinchies, 2019-07-21:
Thanks Bo. I imagine this is the lowest of the low in terms of priorities... 👍

boelle, 2019-07-21:
yep and i'm just sweaping the floor :-)

boelle, 2019-09-24:
@pinchies is the issue still there?

pinchies, 2019-09-24:
I don't know sorry, I haven't been keeping up. Lets remove it for now. Thanks Bo.

closed 2019-09-24 by pinchies

boelle, 2019-09-24:
oki doke, we can always reopen later if we need