MarlinFirmware/Marlin #14260: Bad noise on X axis TMC2130

SSkull995, 2019-06-10:
Hi all! 
I hope you could help me with a problem I am facing on my CreatBot. I am experiencing very bad noise only on the x axis, like rattling, but without layer shifting. 
I honestly don't know why this is happening, I am using 950mAh as the current for both x and y, but it is weird because it just happens on one axis. I tried to look at the belt, but it seems pulled enough. 
Do you know what it should be? I am experiencing this only during non-printing movements, so in SpreadCycle, at 105mm/s. During StealthChop everything works fine. 
Thanks a lot to all of you!

SSkull995, 2019-06-10:
Edit
I noticed that the noise occurs when both the x and the y motors work together, like going on a diagonal. Strange(?)

shitcreek, 2019-06-10:
Have you tried swapping cables, motors and stepper drivers? Is there a physical VREF to adjust? I think you might just need to up the current. 

That said: this Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on this page.) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org Marlin Forum
Tom's 3D Forums
Facebook Group "Marlin Firmware"
Facebook Group "Marlin Firmware for 3D Printers"
Marlin Configuration on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

pattonwebz, 2019-06-26:
While it seems this repo isn't the place to seek help with these kinds of things I just wanted to put a note in here and state I too experience this peroblem and I have not yet found any solution. I have explored possibilities with other community members but have not been able to solve it.

Happens with both stepper driver types I have tried so I don't think it's a problem exclusive to any one type. 

Changing the vrefs higher or going back lower did not resolve the issue but did cause others (overheating, missed steps). My current vrefs are correct for my steppers I believe so adjustments on those again may not be where a solution is.

Swapping X and Y stepper cables (and dropping a brand new steppers in both locations) result in the rattle staying on the X as well, it does not switch.

My machine is an Ender 3 with MKS Gen L board but it happened with the old stock board as well before I switched.

After all my troubleshooting I believe my issue is with the hardware of the machine and probably not the software.

SSkull995, 2019-06-26:
Hi @pattonwebz, yeah I started believing this because it makes these noises randomly. I will try to look into the mechanics!

pattonwebz, 2019-07-04:
@Sskull995 do you also have an Ender 3? Or another printer with 2020 extrusions for X?

I just had a bit of a discovery that I can't believe i missed before now. My timing belt is a standard 6mm wide belt. The entry on the extrusion is 5mm. Swapped to a belt that fits and the noise is gone :) Check your belt for wear on the edges and if it has any wear maybe look to getting a smaller belt (I harvested my replacement from an old 2d printer but any 2mm pitch one would do).

closed 2019-07-11 by thinkyhead

SSkull995, 2019-07-11:
Man I forgot to answer

SSkull995, 2019-07-11:
I will try to change it, but how do I check the entry on the extrusion measure?

pattonwebz, 2019-07-11:
I checked mine with a set of calipers. Looking into it most 2020 extrusions on common printers have 5mm entries. The noise on mine has still not returned after my change (your results may vary).

SSkull995, 2019-07-11:
ok perfect! I will check it and let you know about the results! Thanks!