MarlinFirmware/Marlin #14263: [FR] HOMING_BACKOFF for X2 (IDEX)

richialex, 2019-06-10:
Hello. Is it possible to add HOMING_BACKOFF for X2 in IDEX? When using Sensorless homing, the X2 carriage (which is at zero) knocks heavily (About a second) and only then parked.

InsanityAutomation, 2019-06-13:
Sounds like the home position is incorrect. Is this occurring on tool change? If during homing, the sensitivity is likely too low. If tool change then park position is too high.

richialex, 2019-06-13:
> Sounds like the home position is incorrect. Is this occurring on tool change? If during homing, the sensitivity is likely too low. If tool change then park position is too high.

Thanks for the answer! This happens when the carriage is already in the parking position. If, at the same time, you again execute the G28 command, then it has nowhere to go, respectively, the engine cracks skipping steps (Literally 2 - 3) and only after that sensorless homing is triggered. For the X and Y axis, I enabled the HOMING_BACKOFF option, which eliminates this nuisance, but there is no rollback for the X2 axis.

InsanityAutomation, 2019-06-13:
Can you report you're values for X2_MAX_POS, X2_HOME_POS, and hotend_offset HOTEND_OFFSET_X? I dont have a sensorless endstop machine setup with idex atm, but ill see what the difference in positions is and what happens.

richialex, 2019-06-13:
Sorry for my English, I use a translator.

Those values ​​that you ask for:
#if ENABLED (DUAL_X_CARRIAGE)
  #define X1_MIN_POS X_MIN_POS
  #define X1_MAX_POS X_BED_SIZE
  #define X2_MIN_POS 35
  #define X2_MAX_POS 359.6
  #define X2_HOME_DIR 1
  #define X2_HOME_POS X2_MAX_POS

With these options, everything works fine. The printer prints, the tools are parked nicely.

Problem in another situation:
When I do parking and the tool is out of the house position (for example, in the center), the parking goes well, the carriage touches the obstacle and stops. But if the carriage is already in home position, then the tool is already near the obstacle and if you execute the command “Home”, then before performing the sensorless homing engine skips a couple of steps before it realizes that the obstacle has been reached.

For X1 and Y, you can enable HOMING_BACKOFF, which will allow you to take the carriage a short distance after the park and that solves the problem. But for X2 this is not provided.

The fact is that when using ordinary limit switches, marlin knows its condition (it is pressed or not), and when using sensorless homing, it does not know.

richialex, 2019-06-13:
These are HOMING_BACKOFF parameters:

#define HOMING_BACKOFF_MM {5, 5, 0} // (mm) Move away from the endstops after homing

They are only for X, Y, and Z. For X and Y, it is enabled by 5 mm. When parking, the tool moves away from the obstacle and this is enough so that during the next parking, the engine would not skip steps and sensorless homing would work correctly.

boelle, 2019-07-21:
we need to find out if this is an config error or if @richialex wants a new feature

richialex, 2019-07-21:
Hello! Thank you for your interest.

The whole point of the question will be whether to add the "HOMING_BACKOFF" function for the X2 axis in the IDEX system. In the attached archive configuration files.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3414875/config.zip)

InsanityAutomation, 2019-08-07:
Recently made a change to the homing backoff that may have rectified this situation. Can you test with a current snapshot?

InsanityAutomation, 2019-08-07:
> 
> 
> we need to find out if this is an config error or if @richialex wants a new feature

More like an extension to a feature, that i think overlapped with a bugfix on the same feature I did not long ago.

sjasonsmith, 2020-09-11:
@richialex are you still around? @InsanityAutomation, since they didn't respond to you anytime in the last year, I'm assuming they are not. Do you consider this to be resolved?