MarlinFirmware/Marlin #14273: [FR] TFT screen integration

Gerggg, 2019-06-12:
Hello, Marlin screens are very rich but not very beautiful. while the TFT screens are beautiful but poor in function ...
I believe that the integration of TFT screens is currently done with Gcode commands via serial port, it should be possible to make the integration between the two more complete.
Is it possible to integrate all of Marlin's functions in TFT displays?
thank you for your reply

InsanityAutomation, 2019-06-12:
This is exactly what Extensible UI is designed for. Its currently in use by Lulzbot on the Taz Pro and ive gotten the Creality CR10SPro touchscreen working under it mostly.

Gerggg, 2019-06-12:
thank you for the information but it is a fork of Marlin for the LulzBot 3D printers. This project is not uneven in Marlin.
The idea is the same but I think rather a generic thing that can be integrated to all printers.

InsanityAutomation, 2019-06-12:
Their specific tft integration is yes, but the API is in the main project. A universal touchscreen setup doenst fit in marlin. Theyre external devices with their own firmware. You need to know how the UI is designed to write specific code against them.

Take a look at this here and you should get a better idea of what I mean

https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin_2.0/Marlin/src/lcd/extensible_ui/lib/Creality_DWIN.cpp

Gerggg, 2019-06-13:
Thank you for the information, I see that indeed there are people working for Creality.
I on my side found dianlight working on the biggetech TFT35 to do an integration of Marlin functions. 
I will follow the work of dianlight which makes the generic easier to use.

https://github.com/dianlight/BIGTREETECH-TFT35-V1.2/wiki/Roadmap

Gerggg, 2019-06-18:
It seems that Bigtreetech has already done some of the work by putting the marlin menu in a tft screen (with serial connection).
On the other hand nothing is released yet.

https://www.facebook.com/100021425589762/videos/397007831023400/

marcio-ao, 2019-06-20:
> thank you for the information but it is a fork of Marlin for the LulzBot 3D printers. This project is not uneven in Marlin.

It would be relatively easy to update a plain-vanilla checkout of Marlin with the Lulzbot TAZ Pro interface, all you need to do is copy the directory "Marlin/src/lcd/extensible_ui/lib" from the LulzBot FW into your plain-vanilla Marlin checkout. Then get an FT810 based display, edit the "extensible_ui/lib/config.h" file with the appropriate pins and resolution and enable EXTENSIBLE_UI in "Configuration.h".

The idea behind the EXTENSIBLE_UI is to provide a common interface for different UI implementations, but the source of those implementations would live outside of Marlin as a separate project. The notion of having the Marlin source support all the different displays out-of-box simply doesn't scale that well.

Gerggg, 2019-06-21:
Thank you for your anwer. Now i need to find a FT810 display.

marcio-ao, 2019-06-21:
Two great choices:

https://www.hotmcu.com/43-graphical-ips-lcd-touchscreen-800x480-spi-ft810-p-333.html?cPath=6_16&zenid=0trgrqpdu3000k3nh7u5p7ttu1

https://www.hotmcu.com/5-graphical-ips-lcd-touchscreen-800x480-spi-ft810-p-339.html?cPath=6_16&zenid=0trgrqpdu3000k3nh7u5p7ttu1

I do not recommend the FT800 based ones (much fewer features), nor the capacitative ones (not really supported by our UI).

marcio-ao, 2019-06-21:
We actually began our development with those Haoyu modules, but we eventually made our own board (that adds a speaker, something the Haoyu panels do not have). Our UI is fully compatible with these.

sjasonsmith, 2020-09-11:
A lot of work has happened in the area of displays in the year since this was reported.

We currently have a lot of serial-connected displays which communicate with Marlin in a few different ways. Some use a serial protocol to interact with Marlin (I believe this is the ExtUI @InsanityAutomation) refers to. Others are more directly controlled from Marlin (such as the DWIN displays used by Creality).

We also now have direct-connected TFT displays, which utilize Marlin's color UI. Thanks to efforts from a few people, those are available today on several platforms.

At this point, I would say that a lot of work and improvement has happened, to the point that this request is now too broad to be useful. There are certainly many remaining areas to improve, but those requests (or bugs) need to be more specific to drive progress forward.

I'll go ahead and close this out. If anybody on this thread disagrees with that decision, please reply accordingly.

closed 2020-09-11 by sjasonsmith