MarlinFirmware/Marlin #14274: [BUG] No TMC2208 connection on SKR v1.3

KoffeinKaio, 2019-06-12:
### Description

After removing the jumpers under the Steppers and inserting the UART jumper, TMC2208 is still recognized as "ALL LOW"

### Steps to Reproduce

1: Get TMC2208
2: Solder bridge for UART to PDM pin
3: remove jumper between stepper pins on the SKR2.3 board (not sure about that)
4: insert jumper on UART pins for axis
5: insert stepper driver
6: boot up (assuming all configuration in FW is set)

**Expected behavior:** [What you expect to happen]
UART Connection made

**Actual behavior:** [What actually happens]
UART Connection ALL LOW

#### Additional Information

FW Configuration:
```
  #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3 
  #define X_DRIVER_TYPE  TMC2208
  #define Y_DRIVER_TYPE  TMC2208
  #define Z_DRIVER_TYPE  TMC2208
```

Picture of stepper:
https://frupic.frubar.net/shots/38178.jpg

Picture of board:
https://frupic.frubar.net/shots/38177.jpg

gloomyandy, 2019-06-12:
Are you sure that you have modified the driver boards correctly? In particular with the drivers you are using you may need to bridge the the two centre pins together (I can't see this bridge in your photo) as well as applying the solder bridge to enable UART use. Make sure that you are powering the board from a 12v/24v supply not from USB (from your photo it looks like you have the jumper set to USB).

Given the very many people using this combination (including myself) with Marlin it is very unlikely that there is a bug in Marlin causing your problem. Make sure you are using the latest version of Marlin (not the bigtreetech version), make sure that your platformio install is up to date and that all of the libraries have been updated. 

You will probably get better help for this specific issue from the bigtreetech facebook group or other online forums.

marvelous240, 2019-06-13:
Not a bug.

KoffeinKaio, 2019-06-13:
True sorry all.
I followed a guide which said "you need to short the bridge but not the nc pin on skr 1.3 boards"
As soon as I tested this, it worked. Thank you guys!

closed 2019-06-13 by KoffeinKaio

SirRedZ, 2019-09-03:
@KoffeinKaio  Do you mind clarifying what you did exactly to correct? I have SKR v1.3 w/ 2208's (v3.0).
a.) do I need to solder under each driver for UART mode? (I was told shorting pins 1-2 on the SKR board was the same as the very difficult solder).
b.) Do I need to have any external jumper wires.

Thank you.

theuncle, 2019-09-21:
@SirRedZ I'm in the same setup but I can't get UART to work .
Can you be more specific in which  'Use jumper on pins 1-2 under the stepper' pins are ?

SirRedZ, 2019-09-21:
I apologize.  I WISH I had taken a picture, but I believe this is the correct way to do it.
[image]

I am not 100%, as my board is burried and installed/hidden so I can not verify.

SirRedZ, 2019-09-21:
Note: Powering up without steppers/endstops connected it reported no stealth. When I finished install I was happy to see It had worked as expected.

Nin-D-Tendo, 2019-11-28:
@Tollbringer could you provide more information please?
I have the same setup ( skr 1.3 & tmc2208 v3.0 ), but I can't get it working as u did.

Tollbringer, 2019-11-29:
@Nin-D-Tendo 

This was new to me but I don't mind trying to help. Do you feel the issue might be your firmware?

If so, I have included my Ender 3 - SKRv1.3 // 2208 v3.0 (jumpered on mainboard no solder)
This was my 1st time ever compiling Marlin 2.0x

Let me know if this helps you.

[-=SrZ=-]

[SKR 1.3 + TMC 2208v3_Ender3.zip](https://github.com/MarlinFirmware/Marlin/files/3904655/SKR.1.3.%2B.TMC.2208v3_Ender3.zip)