MarlinFirmware/Marlin #14276: [Bugfix 2.0] LCD problem introduced by commit "Move number-to-string functions to libs"

Bergerac56, 2019-06-12:
A bug seems to have been introduced by commit 2a96d4e23ad0090dc371710d079af731136dfe97
Previous versions were working fine.
When choosing to print a file on a graphic LCD, the first line is screwed up. Here what it gives:
[image]

shitcreek, 2019-06-13:
Confirmed bug. I've got the man, @thinkyhead, on it.

tpruvot, 2019-06-13:
unsure if its related to this commit... There is no numbers printed on this screen in theory... 

btw wasn't "Print" supposed to be the default choice on this "question" ?

LinFor, 2019-06-13:
I have same problem with last-days builds of Marlin.

tpruvot, 2019-06-14:
kinda weird bug... This change almost fixes it, but we lose the "p" of print

> --- a/Marlin/src/lcd/menu/menu_sdcard.cpp
> +++ b/Marlin/src/lcd/menu/menu_sdcard.cpp
> @@ -82,10 +82,11 @@ inline void sdcard_start_selected_file() {
>  #if ENABLED(SD_MENU_CONFIRM_START)
> 
>    void menu_sd_confirm() {
> +    const char* msg = _UxGT("Start print");
>      do_select_screen(
>        PSTR(MSG_BUTTON_PRINT), PSTR(MSG_BUTTON_CANCEL),
>        sdcard_start_selected_file, ui.goto_previous_screen,
> -      PSTR(MSG_START_PRINT " "), card.longest_filename(), PSTR("?")
> +      msg, card.longest_filename(), PSTR("?")
>      );
>    }
>

Bergerac56, 2019-06-14:
Just tested your patch. Strange this "missing p" ! 
It seems that the "p" is replaced by the "_" between "start" and "print"

tpruvot, 2019-06-15:
interesting, maybe there is a kind of "&nbsp;" char available..

tpruvot, 2019-06-15:
fixed with #14292

Bergerac56, 2019-06-15:
The workaround works fine. Till now, I did not find any side effect somewhere else.

closed 2019-06-15 by Bergerac56