MarlinFirmware/Marlin #14280: [BUG] Custom boot screen (still) not displaying on Fysetec mini RGB (v2.1)

Patag, 2019-06-12:
Hi all,
Sorry in advance if my description is not IAW your rules, but since early May, my custom boot screen doesn't display anymore. It was running fine before.
I've tested some workarounds, especially from @DrDitto, but none of them were OK here. As far as I remember, it was after a reorganizing order of the init processes, but can't retrieve the commit date/number. 
All the best
Michel

shitcreek, 2019-06-12:
We try not to provide technical support here. But if it's a bug then we should get to the bottom of it.
Have you tried rolling back to a commit before May? What are the workarounds that you are referring to?

DrDitto, 2019-06-13:
Thanks a lot chech this https://github.com/DrDitto/Marlin

DrDitto, 2019-06-13:
#14216

Patag, 2019-06-13:
I guess I've found the problem : It sounds like the LCD contrast in not yet initialized when Marlin start displaying custom boot screen. In other words, screen remains black while displaying custom boot screen and comes to normal operation juts before the standard boot screen.
Still digging deep in Marlin code, but not familiar enough with it to identify the root cause quickly.
My first feeling is the contrast parameter is loaded from EEPROM at the wrong moment.
To be continued...

gloomyandy, 2019-06-13:
You may want to check the handling of the neopixel backlights, I have a feeling that they are not turned on until after the splash screens are displayed.

thinkyhead, 2019-06-14:
If you are able to narrow down the exact date when the firmware stopped working properly, that may help us to come up with a quick fix. Bisection is a good trick to find the exact date within 4-5 tries.

Patag, 2019-06-14:
My first results : It's an issue related to Lcd backlight AND Lcd contrast.
- For contrast :

In ultralcd_DOGM.cpp, I added this as a workaround
`void MarlinUI::set_contrast(const int16_t value) {`
`    contrast = constrain(value, LCD_CONTRAST_MIN, LCD_CONTRAST_MAX);`
`    u8g.setContrast(contrast);`
**`    u8g.setContrast(255);`**
`}` being called by `MarlinUI::init_lcd()`
This allows Marlin to display the custom screen but only after a manual reset (button), not after a power on reset, backlight being not initialized in this case. IMHO, it makes clear the `contrast` variable is not correctly initialized at this time.

- For backlight : 

`MarlinUI::init_lcd()` set backlight up only for LCDs with a `LCD_BACKLIGHT_PIN`, but Fysetc mini 2.1 not having one, the backlight is not initialized here. I'm still searching for a way to do that; any help welcome :-)

@thinkyhead : I'm not so sure anymore now I was using this LCD in early May. Sorry for the confusion (in my old 52yo brain). I propose you to give up any search on a eventual regression.

gloomyandy, 2019-06-14:
@Bob-the-Kuhn did a lot of work to get this display up and running. He may have some ideas on the best way to handle the splash screen.

Patag, 2019-06-15:
Hi there,
Here are my last results (not real fixes, only workarounds) : 
In ultralcd_DOGM.cpp,
```
.
.
  #include "../../feature/leds/neopixel.h"
.
.
  #if PIN_EXISTS(LCD_BACKLIGHT) // Enable LCD backlight
    OUT_WRITE(LCD_BACKLIGHT_PIN, HIGH);
  #elif ENABLED(FYSETC_MINI_12864_2_1)
    uint8_t background_color[4] = NEOPIXEL_BKGD_COLOR;
    pixels.setBrightness(NEOPIXEL_BRIGHTNESS); // 0 - 255 range
    pixels.setPixelColor(NEOPIXEL_BKGD_LED_INDEX, pixels.Color(background_color[0], background_color[1], background_color[2], background_color[3]));
    pixels.show();
  #endif
  
  #if HAS_LCD_CONTRAST
    refresh_contrast();
    u8g.setContrast(255);
  #endif
```
I still haven't figured out why `contrast `is not correctly initialized at this point.
Hope it will help for a final fix

gloomyandy, 2019-06-15:
On my system the standard Marlin bootscreen works fine, does it on yours? I'm surprised that it works ok if there was a problem with the contrast setting. I do have a user preset colour enabled though, which I suppose may be changing things? Also on my system it goes through a test sequence (red/green/blue) while the splash screen is displayed. Not sure if that is standard or if I enabled it to be honest.

Patag, 2019-06-15:
May be I wasn't clear enough in my descriptions (sorry, English is not my mother tongue). I'm not talking about standard Marlin splash screen, but the custom one when SHOW_CUSTOM_BOOTSCREEN is defined in configuration.h. OK, it's not an essential feature in a 3D printing system, so let's give up if you want.

gloomyandy, 2019-06-15:
Yes I know which screen you are having problems with. My question was really just to confirm that you are not having any problems with the standard Marlin splash screen. They both get displayed in pretty much the same way, and as far as I can remember, one after the other. So if you don't have the custom screen defined does that Marlin one work? If you do have the custom one defined, what happens? Do either of them display properly?

Patag, 2019-06-15:
Ok, the standard Marlin splash screen works and has always worked as expected. I also have the test sequence enabled and it's only at that time LCD start displaying anything.
Did you really try to activate the custom screen ? If so, I can't understand why you do not encounter the problem.
As sais before, don't put this issue on high priority, I'm not a vendor and this feature isn't a show stopper for a hobbyist like me.
Thanks anyway for your effort.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

DrDitto, 2019-07-24:
y fix the boot logo
i will make pubic the changes tomorrow

Patag, 2019-07-24:
So @boelle , please reopen

DrDitto, 2019-07-28:
https://github.com/DrDitto/Marlin/tree/bugfix-2.0.x

DrDitto, 2019-07-28:
PLease the changes of this fork

DrDitto, 2019-07-28:
It address both  panels mks and fysetc

boelle, 2019-07-28:
you can submit a pull request to marlin 2.0.x ?

closed 2019-09-05 by Patag