MarlinFirmware/Marlin #14281: Disable controller fan option in config

ZhenyaKa, 2019-06-13:
If we have relay to control bed, no need to switch controller fan on

Need to add option to disable this lines when we have relay.

\Marlin\src\feature\controllerfan.cpp
      // #if HAS_HEATED_BED
      //   || thermalManager.temp_bed.soft_pwm_amount > 0
      // #endif

Ketchu13, 2019-07-16:
here is what I changed on my fork to not use the controller fan when the bed is used

https://github.com/MarlinFirmware/Marlin/blob/5535a047d71ffe5e688df280d1c38068648b76a3/Marlin/Configuration_adv.h#L280-L285

to:
```c++
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  //#define CONTROLLER_FAN_PIN -1        // Set a custom pin for the controller fan
  //#define CTRF_COOLHBMOSFET
  #define CONTROLLERFAN_SECS 60          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255        // 255 == full speed
#endif
```

https://github.com/MarlinFirmware/Marlin/blob/5535a047d71ffe5e688df280d1c38068648b76a3/Marlin/src/feature/controllerfan.cpp#L41-L43

to:
```c++
      #if HAS_HEATED_BED && ENABLED(CTRF_COOLHBMOSFET)
        || thermalManager.temp_bed.soft_pwm_amount > 0
      #endif
```

boelle, 2019-07-21:
you will still need the fan to blow air on the stepper drivers

TMC driver do require active cooling

ZhenyaKa, 2019-07-21:
Yes, blow when steppers active, but not active when heating bed.

Optionally to switch cooler off when heating bed only

slowbro, 2021-07-03:
This is a duplicate of #11253.

thisiskeithb, 2021-07-03:
Duplicate of #11253

closed 2021-07-03 by thisiskeithb