MarlinFirmware/Marlin #14282: G29 does not activate mesh levelling on SKR 1.3

hudja, 2019-06-13:
Hi, 

Using G29 does not seem to activate mesh levelling on SKR 1.3. The printer probes normally, but when it starts printing no Z axis correction occurs. It seem like that even adding M420 S1 after G29 does not help.

The only way to get mesh levelling working is by manually probing G29 via terminal and saving results, and then by adding only M420 S1 in the gcode.

Thanks!

reloxx13, 2019-06-13:
- post ur startcode
- upload ur configs

hudja, 2019-06-13:
Start code with working mesh bed levelling:

;start
G28 ; Home all axes
;G29 ; bltouch test
M500
M501
M420 S1 ; activate mesh mode
G92 E0 ; Reset Extruder
...

Start code with NON-working mesh bed levelling:
;start
G28 ; Home all axes
G29 ; bltouch test
M500
M501
M420 S1 ; activate mesh mode
G92 E0 ; Reset Extruder
...

Configuration files are in the attached archive.

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3286042/Archive.zip)

InsanityAutomation, 2019-06-13:
same start code for working and non-working?

AnHardt, 2019-06-13:
The '**;**' before the G29 makes the difference.

InsanityAutomation, 2019-06-13:
Ah, i overlooked that.. Is there any terminal output about an invalid mesh when it runs G29? It will not load the mesh if it is not valid and if a probe point was out of range it may leave you without a valid mesh.

hudja, 2019-06-14:
There is nothing wrong with the output of G29, it does not produce any errors, it reports normal mesh, etc, but if your gcode has G29, Z axis is not corrected during the print.

thinkyhead, 2019-06-14:
Don't use `M501` after `G29` or `M420`. In fact, I recommend you don't use `M501` in your start G-code at all.

thinkyhead, 2019-06-14:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-06-14 by thinkyhead