MarlinFirmware/Marlin #14283: [BUG] G33 Not functioning properly

cinealfa, 2019-06-13:
### G33 Doesn't Probe

Initiation of G33 causes a probing failed error 

### Steps to Reproduce

1. issue G33
2.
3. 
**Expected behavior:** G33 begins autocalibration

**Actual behavior:** [What actually happens]
Issue G33
Printhead homes (G28)
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3288129/Archive.zip)

 Printhead makes its way about 2/3 down slows as if to begin probing, then moves 5 or so cm before stopping and showing probing error.
There is no indication in the terminal feed that looks odd to me.

 \
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

cinealfa, 2019-06-13:
Serial output if it helps...
< 2:23:48 PM: G33 Auto Calibrate
< 2:23:48 PM: Checking... AC
< 2:23:48 PM: .Height:285.00  Ex:-1.41  Ey:+0.00  Ez:-6.07  Radius:102.63
< 2:23:48 PM: .               Tx:-0.48  Ty:+1.49  Tz:-1.01
< 2:24:06 PM: Error:Probing failed
< 2:24:06 PM: Correct delta settings with M665 and M666

thinkyhead, 2019-06-14:
CC: @LVD-AC @MarlinFirmware/testers-delta-team

LVD-AC, 2019-06-14:
Check your inititial DELTA_HEIGHT setting.

Bergerac56, 2019-06-16:
I confirm that a possible cause is the initial height. If you modified the usable height of the delta (modification of the position of the end_switches for example) and if the result is that the real height is too far from the initial delta height, probing will fail. 

Try to use a slightly ( a few millimeters) larger value for DELTA_HEIGHT than the manually measured height. Then autocalibrate (and correct if needed)

cinealfa, 2019-06-17:
> Check your inititial DELTA_HEIGHT setting.

The initial height is 300mm which should be more than enough.

cinealfa, 2019-06-17:
The odd thing is I didn't change any configs I just ran an update in Marlin config tool.

LVD-AC, 2019-06-17:
Could also be the steps per mm got set wrong with the update.

BigIronGuru, 2019-06-17:
> Could also be the steps per mm got set wrong with the update.

Plausible. 300mm and 2/3 down and it believes it is around zero, starting from 285. Because the probe routine doesn't go much further down than 0.

cinealfa, 2019-06-18:
Im not sure what is going on I was using the included config in the build. Which up until recently worked fine. Now If i put in some oddball value like 120 it will reach the bed bud the distances are off.

cinealfa, 2019-06-18:
So I related the stock config using the latest build - no change

pdallaire, 2019-06-21:
Hi all, I also have this issue after updating to the latest version. I checked forums for a solution but found nothing that work. I have a log of the auto-level debugging that I include here. I checked the code but since I'm not good at programming, I did not see anything related. I am running Marlin on a BTT Skr V1.3, no hot bed, TMC2100 on X, A4988 on YZE. The Z-probe is a micro-switch on the effector activated by the nozzle bumping the build plate.
[Auto_level_fail.txt](https://github.com/MarlinFirmware/Marlin/files/3315393/Auto_level_fail.txt). Any help would be appreciated.

cinealfa, 2019-06-25:
It is strange all the setting save the type of board are the stock preconfigured setting provided with Marlin. I have to change the steps to get it to work, But work is a relative term. None of the dimensions are correct.

mylife4aiurr, 2019-07-01:
Bugfix 2.0 c28e08c
Bltouch
bilinear bedleveling (also doesn't work)

G33:
auto homes
probe deploys
effector moves to about 10-15mm above printbed with probe deployed
effector stops at this height and errors
Recv: Error:Probing failed

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3347358/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3347359/Configuration.h.txt)

```
Send: M111 S38
Recv: echo:DEBUG:INFO,ERRORS,LEVELING
Recv: ok P31 B3
[...]
Send: G33 P7 V2
Recv: G33 Auto Calibrate
Recv: Checking... AC
Recv: .Height:300.00  Ex:+0.00  Ey:+0.00  Ez:+0.00  Radius:0.00
Recv: .               Tx:+0.00  Ty:+0.00  Tz:+0.00
Recv: reset_bed_level
Recv:   current_position=(0.00, 0.00, nan) : >>> home_delta
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
[...]
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
[...]
Recv: echo:busy: processing
Recv: >>> homeaxis(X)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(X, 450.00, [50.00])
Recv: <<< do_homing_move(X)
Recv: Move Away:
Recv: >>> do_homing_move(X, -5.00, [50.00])
[...]
Recv: echo:busy: processing
Recv: <<< do_homing_move(X)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(X, 10.00, 5.00)
Recv: <<< do_homing_move(X)
Recv: delta_endstop_adj:
Recv: >>> do_homing_move(X, -0.09, [50.00])
Recv: <<< do_homing_move(X)
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Y, 450.00, [50.00])
Recv: <<< do_homing_move(Y)
Recv: Move Away:
Recv: >>> do_homing_move(Y, -5.00, [50.00])
Recv: <<< do_homing_move(Y)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Y, 10.00, 5.00)
[...]
Recv: echo:busy: processing
Recv: <<< do_homing_move(Y)
Recv: delta_endstop_adj:
Recv: >>> do_homing_move(Y, -0.09, [50.00])
Recv: <<< do_homing_move(Y)
Recv: <<< homeaxis(Y)
Recv: >>> homeaxis(Z)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Z, 450.00, [50.00])
Recv: <<< do_homing_move(Z)
Recv: Move Away:
Recv: >>> do_homing_move(Z, -5.00, [50.00])
Recv: <<< do_homing_move(Z)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Z, 10.00, 5.00)
[...]
Recv: echo:busy: processing
Recv: <<< do_homing_move(Z)
Recv: delta_endstop_adj:
Recv: >>> do_homing_move(Z, -0.09, [50.00])
Recv: <<< do_homing_move(Z)
Recv: <<< homeaxis(Z)
Recv: >>> set_axis_is_at_home(X)
Recv:   current_position=(0.00, 0.00, 0.00) :
Recv: <<< set_axis_is_at_home(X)
Recv: >>> set_axis_is_at_home(Y)
Recv:   current_position=(0.00, 0.00, 0.00) :
Recv: <<< set_axis_is_at_home(Y)
Recv: >>> set_axis_is_at_home(Z)
Recv:   current_position=(0.00, 0.00, 300.00) :
Recv: <<< set_axis_is_at_home(Z)
Recv:   current_position=(0.00, 0.00, 300.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 300.00) : <<< home_delta
Recv: >>> probe_pt(0.00, 0.00, raise, 0, nozzle_relative)
Recv:   current_position=(0.00, 0.00, 300.00) :
Recv: >>> do_blocking_move_to(0.00, 0.00, 276.90)
Recv:   destination=(0.00, 0.00, 300.00) : set_destination_from_current
Recv:   destination=(0.00, 0.00, 276.90) : prepare_uninterpolated_move_to_destination
Recv:   current_position=(0.00, 0.00, 276.90) : zone border move
Recv:   current_position=(0.00, 0.00, 276.90) : xy move
Recv: <<< do_blocking_move_to
Recv:   current_position=(0.00, 0.00, 276.90) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(15.00)
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to(0.00, 0.00, 276.90)
Recv:   destination=(0.00, 0.00, 276.90) : set_destination_from_current
Recv:   current_position=(0.00, 0.00, 276.90) : xy move
Recv: <<< do_blocking_move_to
Recv:   current_position=(0.00, 0.00, 276.90) : >>> run_z_probe
Recv:   current_position=(0.00, 0.00, 276.90) : >>> do_probe_move
Recv: >>> do_blocking_move_to(0.00, 0.00, 20.00)
Recv:   destination=(0.00, 0.00, 276.90) : set_destination_from_current
Recv:   current_position=(0.00, 0.00, 276.90) : xy move
Recv:   destination=(0.00, 0.00, 20.00) : prepare_uninterpolated_move_to_destination
Recv:   current_position=(0.00, 0.00, 20.00) : z lower move
Recv: <<< do_blocking_move_to
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv:   current_position=(0.00, 0.00, nan) : sync_plan_position
Recv:   current_position=(0.00, 0.00, nan) : <<< do_probe_move
Recv:   current_position=(0.00, 0.00, nan) : >>> do_probe_move
Recv: >>> do_blocking_move_to(0.00, 0.00, -3.00)
Recv:   destination=(0.00, 0.00, nan) : set_destination_from_current
Recv:   current_position=(0.00, 0.00, nan) : xy move
Recv: <<< do_blocking_move_to
Recv:   current_position=(0.00, 0.00, nan) : sync_plan_position
Recv:   current_position=(0.00, 0.00, nan) : <<< do_probe_move
Recv: SLOW Probe fail!
Recv:   current_position=(0.00, 0.00, nan) : <<< run_z_probe
Recv: >>> do_blocking_move_to(0.00, 0.00, nan)
Recv:   destination=(0.00, 0.00, nan) : set_destination_from_current
Recv:   current_position=(0.00, 0.00, nan) : xy move
Recv: <<< do_blocking_move_to
Recv:   current_position=(0.00, 0.00, nan) : set_probe_deployed
Recv: deploy: 0
Recv: do_probe_raise(15.00)
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv: >>> do_blocking_move_to(0.00, 0.00, nan)
Recv:   destination=(0.00, 0.00, nan) : set_destination_from_current
Recv:   current_position=(0.00, 0.00, nan) : xy move
Recv: <<< do_blocking_move_to
Recv: Error:Probing failed
Changing monitoring state from "Operational" to "Error: Probing failed"
Changing monitoring state from "Error: Probing failed" to "Offline (Error: Probing failed)"
Connection closed, closing down monitor
```

@LVD-AC before at least g33 worked but g29 (bi-linear) did not, now both dont work.

boelle, 2019-09-24:
@cinealfa is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle