MarlinFirmware/Marlin #14284: [FR] Allow "Over shoot" of XY axis during ABL for Offset Probe And Factor glass size

acorrow, 2019-06-13:
### Description

When configuring the Firmware, if you define probing locations beyond the bed's dimensions, you get a Sanity Check error. Rightfully so, but when using a BLTouch (or other probes I'm sure) the offset and configuration mean that you can't probe the very far edge of the bed (CR-10 for reference on size/dimensions).

It get's up as close as it can, which makes sense. However, there is an extra 5MM on the glass X and 15 on the Y, combined with the offset for the BLTouch (42 mm), this leave a heafty amount of the bed that doesn't get probed. Also, there is about 10 MM of "overshoot" play on the X carriage on this machine. The prob could use that extra space to probe further to the right,

Summary:

Request that the firmware allow for "Overshoot" distance (configurable) to allow the X/Y carriage to travel extra to allow further probing

AND

Factor in ACTUAL glass/bed size so that you can offset PROPERLY in BOTH directions. For example:
ONLY XY here:

Bed Size = 235/305 (GLASS SIZE)
Print Area = 220/300
Probe Offset = 42
Min Distance from Edge = 5MM
Grid Points = 4 (16 total)

Probe locations (as shown in the firmware code) are based on start and end. So the first row (X only) would be at points: 5, A, B, 295 (offsetting 5 from each edge, and dynamically choosing the mid points)

This means, with a 42MM offset probe, the TOOL HEAD will actually move to: 47, A, B, 300

This is because it will only ever travel the full distance it's configured to in the FW (300.. thanks sanity check) putting the actual grid points (physical spots the probe hits) at:
5, 66, 196, 258 (Rounding)

While that's a pretty good approximation, with just the 3 extra numbers of the ACTUAL bed size (x, y) and the X "overshoot" allowance (lets call it 10 MM) you could easily calculate the MIDDLE of the ACTUAL bed, figure out (with overshoot) how far you can go to each edge, and INCREASE the distance from the edge as needed. So if the closest (wiht over shoot) you can get to the ACTUAL edge of the glass, is (arbitrary) 30MM, then both ends of the X axis shoudl be inset to that distance, so the points you want to measure are then (again only X)

30, A, B, 270

The toolhead would have to move to:
72, A, B, 312
to make the probe hit those points.  

This would create a more accurate reading assuming you are printing from the middle of the bed.

### Steps to Reproduce

1: Configure your firmware to have a bed slightly smaller than the actual glass
2: Configure ABL with a BLTouch on a CR10 (or any LEFT/RIGHT offset probe)

Expected: Ability to configure glass size and overshoot, to allow more accurate/centered reading of bed level

Actual: Bed measures the points it can reach based on the size of the defined build area, not the glass, and can't travel further than the build volume that is configured creating an inaccurate/incomplete measurement of the build area.

#### Additional Information

You can also just adjust the first point based on the reachable distance to create a "more centered" approach, which is what I will attempt to do in my local build. But this would be a great feature to have.

thisiskeithb, 2019-06-14:
Have you tried setting your `X_MAX_POS` and `Y_MAX_POS` to beyond the bed size? That allow you to reach outside your standard bed size (and probe more area).

I toyed around with those values on my Tornado & Ender-3 to probe as much of the bed from edge to edge as possible, but ended up turning up `GRID_MAX_POINTS_X` (and Y) to 7 and enabled `EXTRAPOLATE_BEYOND_GRID` instead and have been quiet happy with my first layer.

Roxy-3D, 2019-06-14:
@acorrow     The issue you are talking about is real.   But it comes down to what is best for the user.

Most people want full use of their entire print bed.     If you don't care about that...   Just use the INSET options and your probe will be able to reach everywhere in the 'active' bed area.

Most people would rather have the full area of their print bed available.   Because the Z-Probes are typically offset from the nozzle that means there will be some parts of the bed that the probe can't reach.

All of the mesh based systems allow you to use M421 to set a mesh point to the correct value.   If your Z-Probe can't measure an edge point for you...  you can manually set it using M421.    And of course UBL allows you to fill in unmeasured edge points with a G29 P3 command to 'reasonable' values.    Also G26 will produce a pattern showing how well your mesh describes the bed's topology.  

I would encourage you to try these ideas and see how well they work for you.

sjasonsmith, 2020-09-11:
I believe that this request can be handled currently in Marlin 2.0.

Code involving probe offsets and probing boundaries has been extensively reworked since this request was created. When properly configured, the nozzle should be able to leave the printable area to probe to the edge of the bed (or printable area).

I am closing this request, since this should already be possible. If you continue to have issues related to this, please visit one of the resources below and start a conversation. I personally frequent the Discord server.

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) for whichever specific issue remains.

closed 2020-09-11 by sjasonsmith