MarlinFirmware/Marlin #14288: [FR] (feature request title)

ats3788, 2019-06-14:
### Description

### Steps to Reproduce

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ats3788, 2019-06-14:
Ist it possible to print on your LCD Screen the G Code what the printer is working on. like a debug to know where is a problem

marcio-ao, 2019-07-24:
Would you prefer to see a line number or the actual GCODE command?

AnHardt, 2019-07-24:
There are at least two places Marlin is **actually** working on.
The one is the last read g-code line. That's easy.
The other is the actually printed code - where the last planner buffer line was produced from. That would be much harder - but is likely to be much more interesting.

EvilGremlin, 2019-07-24:
It's going way too fast to be of use with LCD. Console output is muuuuch better. It would be good to see line number of last processed gcode before fail though.

ramon-angosto, 2021-12-16:
Any advances on this subject? I'm also interested in this feature.

ellensp, 2021-12-16:
depending on what is happening, the machine could be processing 100's of gcode lines per second. It is unrealistic to print all of them to the LCD.

It would be far better to sprinkle the gcode area your looking at with M117 gcode. https://marlinfw.org/docs/gcode/M117.html
To provide information on where it is at and what it is doing

ramon-angosto, 2021-12-16:
My idea is to print the line number not the entire line, as if it was another parameter like x position. Ideally I would like to send it over the serial port. I think this can be done with the [command M118](https://marlinfw.org/docs/gcode/M118.html). But at least my problem, is how to acces the line number of the Gcode command currently printing. Because if i understand it right [M110](https://marlinfw.org/docs/gcode/M110.html) is used to set a new value for the current line of GCode and not to get the current one.