MarlinFirmware/Marlin #14289: [BUG - Marlin 2.0] Can't compile and build, TMCStepper library missing?

klcjr89, 2019-06-14:
### Description

I am unable to compile and build Marlin 2.0 for a few days now. I was able to compile and build several days ago, but now I get the following errors and I have no idea how to solve it. I'm using macOS. I've searched my entire system and can't find the older <TMCStepper> 0.3.1 that is indicated in the Dependency graph below.

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1769; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x > NXP LPC1769
HARDWARE: LPC1769 120MHz 31.80KB RAM (464KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(strict)
Collected 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.3.1
|-- <Adafruit NeoPixel> 1.1.8

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pioenvs/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pioenvs/LPC1769/src/src/Marlin.cpp.o
Compiling .pioenvs/LPC1769/src/src/feature/babystep.cpp.o
Compiling .pioenvs/LPC1769/src/src/feature/bedlevel/ubl/ubl_G29.cpp.o
Compiling .pioenvs/LPC1769/src/src/feature/bedlevel/ubl/ubl_motion.cpp.o
Compiling .pioenvs/LPC1769/src/src/feature/mixing.cpp.o
Compiling .pioenvs/LPC1769/src/src/feature/pause.cpp.o
Compiling .pioenvs/LPC1769/src/src/feature/power.cpp.o
Compiling .pioenvs/LPC1769/src/src/feature/power_loss_recovery.cpp.o
In file included from Marlin/src/module/stepper.h:221,
from Marlin/src/Marlin.cpp:37:
Marlin/src/module/stepper_indirection.h:62:6: error: #error "Update TMCStepper library to 0.4.1 or newer."
#error "Update TMCStepper library to 0.4.1 or newer."

Patag, 2019-06-14:
Just delete .piolibdeps folder, perform a "clean" command and recompile.
And make sure  TMCStepper@<1.0.0 is defined in your platformio.ini.

klcjr89, 2019-06-14:
@Patag That did the trick! Unfortunately the `piolibdeps` folder was hidden so I had to find it first before I could delete it.

closed 2019-06-14 by klcjr89