MarlinFirmware/Marlin #14295: Can't compile 2.0 for Rumba32 in PIO only

MarkMan0, 2019-06-15:
Cloned Marlin, opened in PIO and compiled with the default settings, everything fine.

Then, per [this](https://github.com/Aus3D/RUMBA32/wiki/Getting-RUMBA32-Up-&-Running#arduino-setup) tutorial I changed the board to Rumba32, Serial port to -1 in Config.h, and selected STM32F4 as default in platformio.ini.

Couldn't compile, error:
`IWDG_HandleTypeDef doesn't name a type` in file _watchdog_STM32F4.h_, line 24.

I went ahead and compiled in Arduino IDE, after installing the STM32Cores package, and it worked, without a problem.

After this I tried to disable the watchdog in Configuration_adv.h, but it didnt help.

Lastly I found [this](https://docs.platformio.org/en/latest/boards/ststm32/rumba32_f446ve.html?utm_source=platformio&utm_medium=piohome), so I changed the board in the .ini file to `rumba32_f446ve`.
It solved the previous errors, but now I have a new one:
`SerialUSB was not declared` in _HAL.h_ line 57.

Please note, that it compiles fine in the Arduino IDE.

MarkMan0, 2019-06-15:
So the files, where the define is used, are the following:
_Marlin\src\HAL\shared\backtrace\**backtrace.cpp**_, on line 39, function `UnwReportOut`. It tries to use SERIAL_CHAR, but it can't.

So I've found that the only definition of SerialUSB is in HAL.cpp, which is wrapped in an `ifdef __STM32F1__`. Since the Rumba uses STM32F4, I think it skips this file, so SerialUSB is not defined. 

What I don't understand is, how can the Arduino IDE compile it.

wesk-ms, 2019-06-16:
Not a Marlin problem.  It's an STM32 conf file issue.  The watchdog module isn't enabled in your STM32 Core.  You need to uncomment this line in your stm32f4xx_hal_conf.h file:
`#define HAL_IWDG_MODULE_ENABLED`

I don't use Platform IO at all (IDE or VS Code plugin).  I simply use VS Code and the Arduino plugin to build Marlin, so I'm not 100% sure where your variant folder lives.  In my case the stm32f4xx_conf.h file is here:
        Users\<username>\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.5.0\variants\RUMBA32_F446VE

I built my own Rumba32 board back in February and have a solid build environment if you need any additional help.  If you try to use a BLTouch with it you'll need to tweak the timer setup in the PeripheralPins.c file as well.  Lastly, all issues that I had with getting Marlin to compile for this board were stm32 core related...nothing on the Marlin side has been an issue.

You picked an awesome mainboard.  I think it's simply the best board for 3D printers...period.  Good luck!

boelle, 2019-09-24:
@MarkMan0 is the issue still there? if so please also attach your config files

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle