MarlinFirmware/Marlin #14296: [FR] FYSETC LCD/TFT

guestisp, 2019-06-15:
Is FYSETC TFT/LCD supported ? They have a custom marlin fork with LCD support on it, i did a simple search on vanilla marlin and found nothing.

Any plans to add ? Or any change to merge a PR, if any ? Maybe (in the far future) I could work on this if there are change for an official merge

https://github.com/FYSETC/Marlin-F6

gloomyandy, 2019-06-15:
I think you need to provide a link to the actual TFT/LCD that you are trying to use. There are several FYSETC displays that are support with bugfix-2.0.x Marlin.

guestisp, 2019-06-15:
This: https://github.com/FYSETC/FYSTLCD

gloomyandy, 2019-06-15:
Hmm well the basic device really just needs to be hooked up to the Marlin serial port (like many other touch screen displays), but looking at the fork it looks like that unlike other TFT controllers this one does not simply use gcode to communicate with Marlin, it looks like they have added extra responses into Marlin for various cases. I'm not sure if a PR adding that sort of code would really be welcome. You may be able to add support using something like the extUI interface (but I don't know much about that), my understanding is that this interface was created with devices like this in mind. Really the right way to do this would be to get FYSETC to add support for their device to Marlin using that interface (or something like it), but I'm not sure how easy it would be to do that. If you don't do that then anytime they update their firmware someone would need to work out what has changed and how to support the change in Marlin, which is not ideal really.

guestisp, 2019-06-15:
Do you know other touch screens supported by Marlin other than the bigtreetech ones? (I really hate they interface)

gloomyandy, 2019-06-15:
As I said above there really is not any "support" in Marlin for these devices, rather they work with standard Marlin using a gcode interface, I think for instance that there is an MKS device that does the same, but I've not used any of them. The problem with all of these devices is that they look pretty but don't give you access to all of the Marlin features that are available through the rather ugly (but functional) simple LCD interfaces.

DavidThijs, 2019-06-15:
The MKS TFT 3.5" v4.0 is working fine on a MKS GEN L 1.0 board and also on an SKR 1.1. I have it running alongside the 128x64 lcd. On my MKS GEN L board, I use it as the sd card reader as my graphic display didn't had one onboard. On the other mainboard, I have tried it and it works but I don't actually use it anymore. The nice thing about the touchscreen is that it shows a pretty progress indicator. Not tried to add buttons for fine steps but the standard ones do work too.

guestisp, 2019-06-15:
Is the firmware opensource ? i'm trying to figure out how to add/customize screens and sent gcode  but without luck

DavidThijs, 2019-06-30:
@[guestisp] Sorry, I didn't tried that yet. I wouldn't call the firmware open source, it's just configurable to my knowledge.

DavidThijs, 2019-06-30:
@[guestisp] Correct that, it's available here: https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/tree/master/firmware
There is also the source for the smaller TFT24

tpruvot, 2019-06-30:
interesting, seems the same panel used in the recent Alfawise U30 pro

https://drive.google.com/file/d/1-oLa0Nzx4_EWl_FdJ4RLKZeQP8hGnsBV/view?usp=sharing