MarlinFirmware/Marlin #14297: Printer system stopped while starting printing[BUG] (short description)

lucasvelosodesantana, 2019-06-16:
Hello, 

I'm a beginner in Marlin. I'm building my first project in Marlin, it consists in a 3D printer that uses two liquids that solidify once they're together. In this project I need to use 4 stepper motors (One fo each axis and one for the extruder) and 3 endstops. The structure has already been build. All the motors and axes and stepper motors have been tested. All the structure works. I tried to remove the `PREVENT_COLD_EXTRUSION` from the code. Everytime i try to start a printing process the printer starts the process and stops, displaying the following message:

Print started at: 17:17:39
echo:Unknown command: "G21"
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:Heating failed, system stopped! Heater_ID: 0
[ERROR] Error:Heating failed, system stopped! Heater_ID: 0

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

These are my configuration and configuration_adv files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3293669/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3293670/Configuration_adv.zip)

shitcreek, 2019-08-16:
Not sure why it's producing that error, but you need to enable 'INCH_MODE_SUPPORT' to use either G20/G21

jimmyeao, 2019-09-17:
I can confirm the same behaviour on an SKR1.3 with TMC2209s. Every 2nd print or so I get this

shitcreek, 2019-09-17:
@jimmyeao need more info: what version of marlin are you running and maybe provide your config? since your error is different I'd start a new issue. @lucasvelosodesantana seems to have abandoned this post.

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle